ghdl/ghdl-yosys-plugin #117: No longer seems to build with newest Yosys?

fl4shk, 2020-05-12:
It doesn't look like ghdl-yosys-plugin is compatible with newest yosys.

Here is the result of me running `make`.

[no_more_build.txt](https://github.com/ghdl/ghdl-yosys-plugin/files/4616591/no_more_build.txt)

tgingold, 2020-05-12:
No, something went wrong in your environment.
In particular `struct logic_32` is the first type defined in `ghdlsynth.h`.  Please check the first lines of this file.  Maybe it wasn't correctly installed.

fl4shk, 2020-05-12:
It seems that perhaps there's something wrong with the yosys I installed with pacman (I'm on Arch Linux).

I'll try building yosys myself and see if that fixes it.

fl4shk, 2020-05-12:
No, wait, I misunderstood what you were saying.  Perhaps it's a problem with my GHDL installation.

closed 2020-05-17 by tgingold

flokli, 2020-06-20:
Just found this as well - it seems the current GHDL release doesn't yet include https://github.com/ghdl/ghdl/commit/c44197a646adae880e0a8e0382dcde64b7aeec35#diff-40994b073e253eeba85e6147897961b7, which is required to compile it with yosys master.

tmeissner, 2020-06-20:
I can compile current master of Yosys, ghdl-yosys-plugin and GHDL without problems. I do that sometimes 2 or 3 times a week to get the most recent features.

tgingold, 2020-06-20:
Indeed, you need master of ghdl to build with Yosys.