ghdl/ghdl-yosys-plugin #120: Improve conversion of records

ozbenh, 2020-05-17:
Hi Tristan !

So I'm trying to use a ghdl+yosys conversion of microwatt to verilog in order to be able to simulate it in the LiteX environment.

The end result isn't quite working, but debugging this is extremely tedious. Microwatt (ab)uses record types for everything under the sun, and having them turned into single verilog vectors makes it extremely hard to parse the wave output.

Would it be realistic to have a mode for ghdl to instead break them up into separate signals whose name is some kind of concatenation of the record name and the field name ? It might generate very large names, so we might want to use a partial record name and some number to dedup but it would still be a lot more useable in sim than what we have now !

tgingold, 2020-05-17:
Well, that's an interesting topic.

That was also my initial plan: synthesis records as a group of wires.
But that doesn't fly well.  Records would become special cases for ports and subprogram arguments.  And if you consider array of records, that would become very difficult.

So I prefer to check if an easiest solution would be possible.

Maybe ghdl could generate a type map to indicate a waveform viewer (gtkwave ?) how to interprete the signals.  The viewer would then split the signal.

Maybe LiteX could also generate vhdl ?

Maybe we should consider writing a `write_vhdl` command to yosys ?  (there is already an issue for that).

tgingold, 2020-05-18:
Do not hesitate to give your opinion on this issue!  That would be useful.

ozbenh, 2020-05-18:
I don't have a strong opinion... either of the above works, whatever is the most realistic to implement :-)

As for LiteX generating GHDL, that's a migen issue, I don't know that much about it either other than it's looks both fantastic in theory and ... really hard to read in practice (at leat for me) ;-)

But as you said, it might be doable instead to have yosys turn the verilog into vhdl.