ghdl/ghdl-yosys-plugin #126: Testing on some designs from Opencores

Chopper455, 2020-06-18:
Used msys2 32bit in Windows 7 to integrate everything together.
I've compiled ghdl 81905a8c.
Then used ghdl-yosys-plugin 0b687cd.
In-compiled it directly into the yosys 334ec5fa.
[fails.zip](https://github.com/ghdl/ghdl-yosys-plugin/files/4797482/fails.zip)
I hope this will help you to make this piece of work even better:)

Used several designs from opencores.org, to check if they really can be synthesized used Quartus Prime Lite Edition 16.1.
Some designs where ghdl falls with "GHDL Bug occurred" weren't checked in Quartus.

[https://opencores.org/projects/8b10b_encdec](url)
Command:
`ghdl ./8b10_enc.vhd -e`
Result:
> 1. Executing GHDL.
note: top entity is "enc_8b10b"
>
> ******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-755-g81905a8c-dirty) [Dunoon edition]
Compiled with unknown compiler version
Target: i686-w64-mingw32
C:\Programs\FreeEDA\ghdl_git\!fails\8b10b_encdec\
Command line:
C:\Programs\FreeEDA\yosys-master\yosys_bin\bin\yosys.exe
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists-memories.adb:2578
>******************************************************************
>ERROR: vhdl import failed.

[https://opencores.org/projects/802154phycore](url)
In Quartus used "ieee_802_15_4_phy" top-module, but in ghdl used a lower level module.
Command:
`ghdl -fsynopsys -fexplicit ./chip_gen.vhd ./upsampler.vhd ./tx_fir.vhd ./tx_core.vhd -e`
Result:

> 1. Executing GHDL.
note: top entity is "tx_core"
./chip_gen.vhd:65:62: unhandled predefined IEEE operator "rol"
../../src/ieee/v93/numeric_std.vhdl:645:12:  declared here
./chip_gen.vhd:66:62: unhandled predefined IEEE operator "rol"
../../src/ieee/v93/numeric_std.vhdl:645:12:  declared here
./tx_fir.vhd:82:47: unhandled (static) function: IIR_PREDEFINED_IEEE_NUMERIC_STD_RESIZE_SGN_NAT
> 
>******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-755-g81905a8c-dirty) [Dunoon edition]
Compiled with unknown compiler version
Target: i686-w64-mingw32
C:\Programs\FreeEDA\ghdl_git\!fails\802154phycore\
Command line:
C:\Programs\FreeEDA\yosys-master\yosys_bin\bin\yosys.exe
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : synth-expr.adb:793 access check failed
> ******************************************************************
>ERROR: vhdl import failed.

[https://opencores.org/projects/avs_aes](url)
Quartus compiled fine the AES_CORE top-module.
Command:
`ghdl --work=avs_aes_lib ./addroundkey.vhd ./aes_core.vhd ./aes_fsm_decrypt.vhd ./aes_fsm_encrypt.vhd ./avs_aes.vhd ./avs_aes_pkg.vhd ./keyexpansionV2.vhd ./memory_word.vhd ./mixcol.vhd ./mixcol_fwd.vhd ./mixcol_inv.vhd  ./mux3.vhd ./sbox.vhd ./sbox_arch1.vhd ./shiftrow.vhd ./shiftrow_fwd.vhd ./shiftrow_inv.vhd -e
`
Result:
>1. Executing GHDL.
note: top entity is "avs_aes"
./aes_core.vhd:122:24:warning: instance "multiplex" of component "mux2" is not bound [-Wbinding]
./aes_core.vhd:75:14:warning: (in default configuration of aes_core(arch1))
./aes_core.vhd:133:16:warning: instance "keyindexmux" of component "mux2" is not bound [-Wbinding]
[... 198 lines cut ...]

[https://opencores.org/projects/viterbi_decoder_axi4s](url)
Quartus sythesized with "dec_viterbi" as a top-module, maybe I just set library name incorrectly.
Command:
`ghdl -fsynopsys --work=dec_viterbi ./acs.vhd ./axi4s_buffer.vhd ./branch_distance.vhd ./dec_viterbi.vhd ./generic_sp_ram.vhd ./pkg_components.vhd ./pkg_helper.vhd ./pkg_param.vhd ./pkg_param_derived.vhd ./pkg_trellis.vhd ./pkg_types.vhd ./ram_ctrl.vhd ./recursion.vhd ./reorder.vhd ./traceback.vhd -e
`
Result:
>1. Executing GHDL.
./dec_viterbi.vhd:29:8: identifier "dec_viterbi" already used for a declaration
*libraries*:1:1: previous declaration: library "dec_viterbi"
./dec_viterbi.vhd:29:8: identifier "dec_viterbi" already used for a declaration
*libraries*:1:1: previous declaration: library "dec_viterbi"
./dec_viterbi.vhd:321:33: component name expected, found if generate statement
note: top entity is "dec_viterbi"
ERROR: vhdl import failed.

_renamed the entity, tried again_
Command:
`ghdl -fsynopsys --work=dec_viterbi ./acs.vhd ./axi4s_buffer.vhd ./branch_distance.vhd ./dec_viterbi_rename.vhd ./generic_sp_ram.vhd ./pkg_components.vhd ./pkg_helper.vhd ./pkg_param.vhd ./pkg_param_derived.vhd ./pkg_trellis.vhd ./pkg_types.vhd ./ram_ctrl.vhd ./recursion.vhd ./reorder.vhd ./traceback.vhd -e
`
Result:
>1. Executing GHDL.
./dec_viterbi_rename.vhd:321:33: component name expected, found if generate statement
note: top entity is "dec_viterbi_rename"
ERROR: vhdl import failed.

eine, 2020-06-18:
> In-compiled it directly into the yosys 334ec5fa.

Does this mean that you tried https://github.com/ghdl/ghdl-yosys-plugin#build-as-part-of-yosys-not-recommended instead of https://github.com/ghdl/ghdl-yosys-plugin#build-as-a-module-shared-library?

tgingold, 2020-06-18:
I will first fix the missing features, but:
- I am not sure what to do with the invalid vhdl.
- I think it is better to create one issue per project as it is easier to track progress and status.

Chopper455, 2020-06-19:
yes, i've tried #build-as-part-of-yosys-not-recommended because i couldn't compile it as module because of missing yosys symbols during linking, it was easier for me just to integrate it in yosys to make symbols visible.
In the future i'll create separate issues, thank you.

Chopper455, 2020-06-19:
i'm not in a VHDL camp so i'm not familiar with lexical/syntax rules in standard, i consulted Quartus to verify them for me.