ghdl/ghdl-yosys-plugin #131: Multiple use of a module

IIupor, 2020-09-08:
**Description**
ghdl or yosys changes module port names to lowercase

**Expected behaviour**
I want to use the module (i.e. entity) mux2_1 in mux4_1 and mux4_1_x2.
Using 

```
ghdl mux2_1.vhd -e mux2_1
ghdl mux4_1.vhd -e mux4_1
```

i got error
```ERROR: Re-definition of module `\ mux2_1 '```

Using
```ghdl mux2_1.vhd mux4_1.vhd -e mux4_1```
i got error
```ERROR: Module `mux2_1 'referenced in module` mux4_1_x2' in cell `mux1 'does not have a port named' Y '.```

To determine the cause of the error, I wrote verilog file with the following command:
```write_verilog ./mux4_1_x2.syn0.v```

It turned out that ghdl or yosys changed the port names of the module to lower case
```
module mux2_1 (s0, d1, d0, y);
  wire _0_;
  wire _1_;
  wire _2_;
  wire _3_;
  input d0;
  input d1;
  input s0;
  output y;
  assign _0_ = ~ s0;
  assign _1_ = d0 & _0_;
  assign _2_ = d1 & s0;
  assign _3_ = _1_ | _2_;
  assign y = _3_;
endmodule
```
How can a module be used in different files?

**How to reproduce?**

```library ieee;
use ieee.std_logic_1164.all;

entity mux2_1 is
port --define inputs and outputs        
(
	S0  :  in std_logic;
	D1  :  in std_logic;
	D0  :  in std_logic;
	Y   :  out  std_logic 
);

end mux2_1;

architecture logic of mux2_1 is
begin
Y <= (D0 and (not S0)) or 
     (D1 and S0) ; 
end logic;

library ieee;
use ieee.std_logic_1164.all;

entity mux4_1 is
port --define inputs and outputs        
(
	S1  :  in std_logic;
	S0  :  in std_logic;
	D3  :  in std_logic;
	D2  :  in std_logic;
	D1  :  in std_logic;
	D0  :  in std_logic;
	Y   :  out  std_logic 
);

end mux4_1;

architecture struc of mux4_1 is
	component mux2_1 is
		port(
				S0  :  in std_logic;
				D1  :  in std_logic;
				D0  :  in std_logic;
				Y   :  out  std_logic 
			);
	end component;
    signal out_mux0    : std_logic;
    signal out_mux1    : std_logic;
begin
    mux2_1_0 : mux2_1
      port map (S0    => S0,
                D1    => D1,
                D0	  => D0,
                Y     => out_mux0
                );

    mux2_1_1 : mux2_1
      port map (S0    => S0,
                D1    => D3,
                D0	  => D2,
                Y     => out_mux1
                );

    mux2_1_2 : mux2_1
      port map (S0    => S1,
                D1    => out_mux1,
                D0	  => out_mux0,
                Y     => Y
                );
end struc;

module mux4_1_x2 (out1, out2, data1, data2, sel1, sel2);
    input [3:0] data1;
    input [1:0] data2;

    input [0:1] sel1;
    input sel2;
    output out1;
    output out2;

    mux4_1 mux0 (.S0(sel1[0]), .S1(sel1[1]), .D3(data1[1]), .D2(data1[0]), .D1(data1[1]), .D0(data1[0]), .Y(out1));
    mux2_1 mux1 (.S0(sel2), .D1(data2[1]), .D0(data2[0]), .Y(out2));
endmodule
```

tgingold, 2020-09-08:
VHDL being case insensitive, ghdl only uses lowercase names.  Yosys being verilog is case sensitive and expect the same name.

So, you can either use lowercase ports name in verilog (that should work), or rename the ports once imported (using the command rename in yosys).

I fear there is no magic solution.

IIupor, 2020-09-08:
Thank you for a answer