ghdl/ghdl-yosys-plugin #133: Q: can I mix VHDL with Verilog?

rodrigomelo9, 2020-09-26:
Hi. I am adding support for `ghdl --synth` and `ghdl-yosys-plugin` into [PyFPGA](https://gitlab.com/rodrigomelo9/pyfpga). I have some questions about ghdl-yosys-plugin.

I have four files:
* `blinking.vhdl`
* `examples_pkg.vhdl`: a package which includes `blinking.vhdl` 
* `top.vhdl` and `top.v` the same top level (entity/module called Top).

I am using `ghdl/synth:beta`.

It works (`case 1`):
```
FLAGS="--std=08 -fsynopsys -fexplicit -frelaxed"
$DOCKER_CMD ghdl -a $FLAGS --work=examples ../../hdl/blinking.vhdl
$DOCKER_CMD ghdl -a $FLAGS --work=examples ../../hdl/examples_pkg.vhdl
$DOCKER_CMD ghdl -a $FLAGS ../../hdl/top.vhdl
$DOCKER_CMD yosys -Q -m ghdl -p '
ghdl '"$FLAGS"' Top;
synth_xilinx -family xc7;
write_edif -pvector bra yosys.edif
'
```
It works (`case 2`):
```
$DOCKER_CMD yosys -Q -m ghdl -p '
ghdl '"$FLAGS"' --work=examples ../../hdl/blinking.vhdl ../../hdl/examples_pkg.vhdl ../../hdl/top.vhdl -e Top;
synth_xilinx -family xc7;
write_edif -pvector bra yosys.edif
'
```
**First question:** what happens if I have multiple files in different packages? Must I use only `case 1` in such a case?

I haven't found how to mix languages. So the **Second question:** is it possible?

I tried:
```
$DOCKER_CMD yosys -Q -m ghdl -p '
ghdl '"$FLAGS"' ../../hdl/blinking.vhdl;
read_verilog -defer ../../hdl/top.v;
synth_xilinx -top Top -family xc7;
write_edif -pvector bra yosys.edif
'
1. Executing GHDL.
error: bad unit name '../../hdl/blinking.vhdl'
error: (a unit name is required instead of a filename)
ERROR: vhdl import failed.
```
Also:
```
$DOCKER_CMD ghdl -a $FLAGS --work=examples ../../hdl/blinking.vhdl

$DOCKER_CMD yosys -Q -m ghdl -p '
read_verilog -defer ../../hdl/top.v;
synth_xilinx -top Top -family xc7;
write_edif -pvector bra yosys.edif
'
```
Where Yosys starts the synthesis but:
```
2.4.1. Analyzing design hierarchy..
Top module:  \Top
ERROR: Module `\Blinking' referenced in module `\Top' in cell `\dut' is not part of the design.
```
And other alternatives, without success :-(

**Disclaimer:** I didn't read that it is possible, I am only trying to figure how to implement the most complete support into PyFPGA :-D

rodrigomelo9, 2020-09-26:
Sorry, an extra question. For synthesis, I am using as default options `--std=08 -fsynopsys -fexplicit -frelaxed`. Seems ok? Another recommendation? The idea is to avoid the need for such configurations from PyFPGA.

eine, 2020-09-26:
> what happens if I have multiple files in different packages? Must I use only case 1 in such a case?

Assuming that you mean "logical library" with "packages", I think so. You might do a single call per logical library, tho. Moreover, I think that you might include the top-level file in the yosys command:

```sh
FLAGS="--std=08 -fsynopsys -fexplicit -frelaxed"
$DOCKER_CMD ghdl -a $FLAGS --work=examples ../../hdl/blinking.vhdl ../../hdl/examples_pkg.vhdl
$DOCKER_CMD yosys -Q -m ghdl -p '
ghdl '"$FLAGS"' ../../hdl/top.vhdl -e Top;
synth_xilinx -family xc7;
write_edif -pvector bra yosys.edif
'
```

---

> I haven't found how to mix languages. So the Second question: is it possible?

It is, but:

- The result will be a Yosys verilog, EDIF or JSON output, but not VHDL. @rlee287's #122 is work in progress for adding a VHDL exporter to Yosys. Then, it should be possible to combine VHDL and Verilog, but generate VHDL output.
- I'm not sure about how "automated" it is. See @hackfin's comments in [June 29, 2020 9:15 PM](https://gitter.im/ghdl1/Lobby?at=5efa3dd5c223cc536a42888b):

> At this moment you'd have to create a component (stubs) package from a set of the verilog files you want to include. There are some conversion tricks with iverilog, but it gets ugly when generics come into play.

> I pushed a preliminary 'playground' here: https://github.com/hackfin/hdlplayground containing some notebooks with examples (hit Binder button, wait until it's up, use play button to step through the cells to reproduce)
> Haven't gotten round yet to document the Verilog blackbox inferences in detail, I'm afraid

See also #100, #77 and #46.

---

> For synthesis, I am using as default options --std=08 -fsynopsys -fexplicit -frelaxed. Seems ok? Another recommendation? The idea is to avoid the need for such configurations from PyFPGA.

Strictly, it is not ok precisely because `-fsynopsys -fexplicit -frelaxed` are workarounds for code that is not compliant with the LRM. However, should you want to synthesise vendor IP, then you need them. I would probably group them as an option in PyFPGA, but I would not enable it by default (unless targetting vendor IP only).

rodrigomelo9, 2020-09-26:
> ```shell
> FLAGS="--std=08 -fsynopsys -fexplicit -frelaxed"
> $DOCKER_CMD ghdl -a $FLAGS --work=examples ../../hdl/blinking.vhdl ../../hdl/examples_pkg.vhdl
> $DOCKER_CMD yosys -Q -m ghdl -p '
> ghdl '"$FLAGS"' ../../hdl/top.vhdl -e Top;
> synth_xilinx -family xc7;
> write_edif -pvector bra yosys.edif
> '
> ```

Ok, so the best approach for a general situation seems to be an analysis of each VHDL with a `ghdl -a` command.

> > I haven't found how to mix languages. So the Second question: is it possible?
> 
> It is, but:...

Ok, I see. I was asking particularly with `ghdl-yosys-plugin`. I am not interested in a particular output format (VHDL, Verilog, EDIF, whatever). I am trying to provide the complete flow from VHDL+Verilog to bitstream. I will check the provided alternatives.

> > For synthesis, I am using as default options --std=08 -fsynopsys -fexplicit -frelaxed. Seems ok? Another recommendation? The idea is to avoid the need for such configurations from PyFPGA.
> 
> Strictly, it is not ok precisely because `-fsynopsys -fexplicit -frelaxed` are workarounds for code that is not compliant with the LRM. However, should you want to synthesise vendor IP, then you need them. I would probably group them as an option in PyFPGA, but I would not enable it by default (unless targeting vendor IP only).

Could be problems if enabled by default? (and without the option of being disabled?) I know that hey are workarounds, and I always avoid to use not compliant code in my projects, but the idea is to provide an easy solution for most cases.

Thanks for the info.

eine, 2020-09-26:
> Ok, so the best approach for a general situation seems to be an analysis of each VHDL with a `ghdl -a` command.

I'd say analysing each Library with a command. For instance, if a fusesoc file is used each "fileset" (see e.g. https://github.com/chipsalliance/Cores-SweRVolf/blob/master/swervolf.core#L5) might be compiled with a command.
 
> Ok, I see. I was asking particularly with `ghdl-yosys-plugin`. I am not interested in a particular output format (VHDL, Verilog, EDIF, whatever). I am trying to provide the complete flow from VHDL+Verilog to bitstream. I will check the provided alternatives.

Although I don't know the details of how ghdl-yosys-plugin and Yosys handle it, there are some VHDL features not directly representable in Verilog. Moreover, depending on the "backend" used after Yosys, HDL output might be needed. That is, not all P&R tools support the JSON output from Yosys. Some need BLIF, some other need EDIF (which is not standard), some need HDL.

I understand that you want to focus on nextpnr, and the JSON output is enough. It is good to keep an eye on other alternatives, tho.
 
> Could be problems if enabled by default? (and without the option of being disabled?) I know that hey are workarounds, and I always avoid to use not compliant code in my projects, but the idea is to provide an easy solution for most cases.

It depends on your target audience. If most of the users will depend on e.g. Xilinx IP, then it makes sense to enable it why default. However, if most users are expected to process correct VHDL, they might get strange errors, warnings or bugs which they should not.

The general approach in the VASG and in GHDL is to prioritise correctness over comfort. Therefore, my suggestion is not enabling it by default, but adding a clear note either in the log or in the docs. Changing it should be as simple as modifying a parameter. The importance relies on users being aware of why they are doing it.

rodrigomelo9, 2020-10-11:
I know now how to accomplish it thanks to @eine. Examples [here](https://gitlab.com/rodrigomelo9/pyfpga/-/tree/yosys-nextpnr-icestorm/poc/os-mix) (temporary branch, will be available at master, when merged, [here](https://gitlab.com/rodrigomelo9/pyfpga/-/tree/master/poc/os-mix)). Check `works.sh` to see the working alternatives.

closed 2020-10-11 by rodrigomelo9