ghdl/ghdl-yosys-plugin #134: Something related to deal with packages seems to break the Yosys EDIF generation for ISE

rodrigomelo9, 2020-09-29:
Hi. I am not sure about to post it here or at the Yosys repo, but a first glance, it seems related to deal with VHDL packages. Let me explain.

* I was adding `ghdl-yosys-plugin` support into [PyFPGA](https://gitlab.com/rodrigomelo9/pyfpga).
* I have four examples, where a blinking led is tested on hardware, and only one has problems:
  * Verilog with Yosys, using ISE (Spartan 6) to generate the bitstream, is working with a LX9 microboard.
  * Verilog with Yosys, using Vivado (Zynq 7000) to generate the bitstream, is working with a Zybo.
  * VHDL with `ghdl-yosys-plugin` + Yosys, using Vivado (Zynq 7000) to generate the bitstream, is working with a Zybo.
  * VHDL with `ghdl-yosys-plugin` + Yosys, using ISE (Spartan 6), fails with a complaint about the EDIF format:
```
Command Line: /opt/Xilinx/14.7/ISE_DS/ISE/bin/lin64/unwrapped/ngdbuild -intstyle
ise -dd _ngo -nt timestamp -uc hdl/s6micro.ucf -p xc6slx9-csg324-2 ise.edif
Top.ngd

Executing edif2ngd -quiet "ise.edif" "_ngo/ise.ngo"
WARNING: es_AR:es is not supported as a language.  Using usenglish.
Release 14.7 - edif2ngd P.20131013 (lin64)
Copyright (c) 1995-2013 Xilinx, Inc.  All rights reserved.
ERROR:NgdBuild:202 - On or above line 1277 in file "ise.edif":  EDIF
   asciiCharacter specification has illegal character """.  This likely means
   that the EDIF netlist was improperly written.  Please contact the vendor of
   the program that produced this EDIF.
ERROR:NgdBuild:276 - edif2ngd exited with errors (return code 2).
ERROR:NgdBuild:28 - edif2ngd did not successfully complete.  Please check
   preceding errors for root cause.
Total REAL time to NGDBUILD completion:  2 sec
Total CPU time to NGDBUILD completion:   0 sec

Writing NGDBUILD log file "Top.bld"...
WARNING: es_AR:es is not supported as a language.  Using usenglish.
Release 14.7 - edif2ngd P.20131013 (lin64)
Copyright (c) 1995-2013 Xilinx, Inc.  All rights reserved.
ERROR:NgdBuild:202 - On or above line 1277 in file "ise.edif":  EDIF
```
Here you are [issue.tar.gz](https://github.com/ghdl/ghdl-yosys-plugin/files/5300717/issue.tar.gz), with:
* hdl/*: the sources (a blinking, with a VHDL package, used in a top-level file, and the constraints file)
* `works.sh:` a shell script to generate the project using only blinking.vhdl (without the package and the top-level. It works (no complaints about the EDIF).
* `fails.sh`: similar shell script but using blinking.vhdl + package + top-level. It fails

Again, it could be related to the VHDL packages handling or only a problem with the ISE EDIF generator of Yosys. Let me know what you think, and if I must to report it to Yosys instead.

I tried the conversion from the three VHDLs to one Verilog (using write_verilog instead of write_edif) and it also works.

Regards,
Rodrigo

tgingold, 2020-09-30:
Apparently ISE doesn't like the '%' in the strings.  I have to check whether the '%' is allowed in EDIF.

rodrigomelo9, 2020-09-30:
You can check, of course, but we need to avoid the '%' (if the problem) to use with devices such as Spartan6 :smiley:

tgingold, 2020-10-02:
Can you try with this change:
```
--- a/src/ghdl.cc
+++ b/src/ghdl.cc
@@ -48,7 +48,8 @@ static std::string to_str(Sname name)
                        res = '.' + string(get_cstr(get_sname_suffix(pfx))) + res;
                        break;
                case Sname_Version:
-                       res = '%' + stringf("%u", get_sname_version(pfx)) + res;
+                       //  Use '$' for versions.  '%' is not supported by Xilinx ISE edif2ngc.
+                       res = '$' + stringf("%u", get_sname_version(pfx)) + res;
                        break;
                }
        }
```

rodrigomelo9, 2020-10-02:
It works, ISE and Vivado (working in hardware :-D). Great work @tgingold and thanks @eine for the support (regenerate `ghdl/synth:beta` locally).

commit 6e00f79c (2020-10-02): '%' is not supported by Xilinx ISE edif2ngc. Fix #134

Authored-By: Tristan Gingold <tgingold@free.fr>
closed 2020-10-02 by eine