ghdl/ghdl-yosys-plugin #140: Compilation broken

zipotron, 2020-12-22:
Hello, I am trying to compile the plugin, but I got a compilation error, I have all dependencies installed, I paste the message error down the text. Thanks for this amazing project, I hope we can find a solution and use it.

[zipotron@localhost ghdl-yosys-plugin]$ make
yosys-config --exec --cxx -c --cxxflags -o ghdl.o src/ghdl.cc -fPIC -DYOSYS_ENABLE_GHDL -I/usr/local/include -O
src/ghdl.cc: En la función ‘void import_module(Yosys::RTLIL::Design*, GhdlSynth::Module)’:
src/ghdl.cc:620:59: error: no hay coincidencia para la llamada a ‘(Yosys::hashlib::pool<Yosys::RTLIL::IdString>) (std::string)’
  620 |    module->avail_parameters(to_str(get_param_name(m, idx)));
      |                                                           ^
src/ghdl.cc:925:12: error: ‘struct Yosys::RTLIL::Module’ has no member named ‘addModFloor’
  925 |    module->addModFloor(to_str(iname), IN(0), IN(1), OUT(0), id == Id_Smod);
      |            ^~~~~~~~~~~
make: *** [Makefile:25: ghdl.o] Error 1

eine, 2020-12-22:
Which version of Yosys and ghdl-yosys-plugin are you trying to build?

zipotron, 2020-12-22:
Yosys 0.9 and the ghdl-yosys-plugin is the last code in the git repo

zipotron, 2020-12-22:
And is in a x86_64 machine, ghdl had no compilation issues

Xiretza, 2020-12-22:
You need a recent git version of yosys, 0.9 is quite old (though there are no newer releases).

zipotron, 2020-12-22:
Was that, its working with the last yosys code. Thanks!

closed 2020-12-22 by zipotron