ghdl/ghdl-yosys-plugin #142: make fails

github-4o, 2021-01-31:
```
$ git clone git@github.com:ghdl/ghdl-yosys-plugin.git
Cloning into 'ghdl-yosys-plugin'...
cremote: Enumerating objects: 96, done.
remote: Counting objects: 100% (96/96), done.
remote: Compressing objects: 100% (66/66), done.
remote: Total 1093 (delta 41), reused 74 (delta 27), pack-reused 997
Receiving objects: 100% (1093/1093), 231.28 KiB | 525.00 KiB/s, done.
Resolving deltas: 100% (489/489), done.
$ cd ghdl-yosys-plugin/
$ make
yosys-config --exec --cxx -c --cxxflags -o ghdl.o src/ghdl.cc -fPIC -DYOSYS_ENABLE_GHDL -I/home/pi/repos/ghdl/build/PREFIX/include -O
src/ghdl.cc:1291:18: error: use of undeclared identifier 'get_identifier'
                        nameid_gclk = get_identifier("gclk");
                                      ^
1 error generated.
make: *** [Makefile:25: ghdl.o] Error 1
$ ghdl --version
GHDL 1.0-dev (v0.37.0-1227-g36ae5c1f) [Dunoon edition]
 Compiled with GNAT Version: 8.3.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2020 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
$ yosys --version
Yosys 0.9+3833 (git sha1 b0004911, clang 7.0.1-8+rpi3+deb10u2 -fPIC -Os)```

Xiretza, 2021-01-31:
You'll need a newer version of ghdl, 36ae5c1f is about a month old.

github-4o, 2021-01-31:
ok, checking it

github-4o, 2021-02-01:
```
$ make                                                                                
yosys-config --exec --cxx -c --cxxflags -o ghdl.o src/ghdl.cc -fPIC -DYOSYS_ENABLE_GHDL -I/usr/local/include -O
yosys-config --build ghdl.so ghdl.o -shared /usr/local/lib/libghdl-1_0_0rc3.so -Wl,-rpath,/usr/local/lib/                      
$ make install                                                                   
yosys-config --exec mkdir -p --datdir/plugins
yosys-config --exec cp ghdl.so --datdir/plugins                                                                                
$ ghdl --version                                                                      
GHDL 1.0.0rc3 (1.0.0rc1.r2.g4dc37524) [Dunoon edition]
Compiled with GNAT Version: 8.3.0                                                                                              
llvm code generator                                                                                                           
Written by Tristan Gingold.                                                                                                                               
Copyright (C) 2003 - 2021 Tristan Gingold.                                                                                     
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.                                                    
$ yosys --version                                                                     
Yosys 0.9+3833 (git sha1 b0004911, clang 7.0.1-8+rpi3+deb10u2 -fPIC -Os)
$ git log -1                                                                          
commit cba859cacf8c6631146dbdaa0f297c060b5a68cd (HEAD -> master, origin/master, origin/HEAD)                                   
Author: Tristan Gingold <tgingold@free.fr>
Date:   Mon Jan 25 18:32:14 2021 +0100                                                                                                                                                                                                                            testsuite: add testcase for ghdl/ghdl#1610
```
updating grlib fixed the build

closed 2021-02-01 by github-4o