ghdl/ghdl-yosys-plugin #144: Unable to synthesize large design

monideepbora, 2021-03-15:
Dear author(s),

I have been trying to synthesize the top CPU module of the NEORV32 core using custom cell libraries. However, yosys cant seem to find the cpu entity which has already been analysed. The RTL files and the script are uploaded to https://github.com/monideepbora/neor32_synthesis.git

The commands that are run are in the ghdl/script.sh file and the error I encounter is given below:

```
 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Wolf <claire@symbioticeda.com>          |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+3838 (git sha1 339848b9, gcc 9.3.0 -fPIC -Os)

-- Running command `ghdl neorv32_cpu' --

1. Executing GHDL.
error: cannot find entity or configuration neorv32_cpu
ERROR: vhdl import failed.
```

Please let me know if I am doing anything wrong.

Thanks and best regards,
Monideep

eine, 2021-03-15:
Try `yosys -p 'ghdl --work=neorv32 neorv32_cpu'`.

monideepbora, 2021-03-15:
Hello @eine ,

Thanks for the quick revert. It solves the issue. 

Thanks & best regards,
Monideep