ghdl/ghdl-yosys-plugin #147: GHDL VHDL support

PPlinux, 2021-03-30:
Dear,

is it possible to synthesize the AX8 design written in vhdl https://opencores.org/projects/ax8 for fpga?

Is there a ghdl - yosys ... design flow available for Windows computers or is it only supported on Linux ?

Hoping for positive answer or guidance to try it out my self.

Greetings,

Patrick

umarcor, 2021-03-30:
> Is there a ghdl - yosys ... design flow available for Windows computers or is it only supported on Linux ?

There is! GHDL and Yosys with ghdl-yosys-plugin are available in MSYS2. See https://hdl.github.io/MINGW-packages/ and https://packages.msys2.org/group/mingw-w64-x86_64-eda. For installing the minimal tools, do:

```
pacman -S mingw-w64-x86_64-yosys
```

For installing other EDA packages, do:

```
pacman -S mingw-w64-x86_64-eda
```

> is it possible to synthesize the AX8 design written in vhdl https://opencores.org/projects/ax8 for fpga?

No better answer than trying :smile:

First, GHDL synthesis only:

```
ghdl synth *.vhd -e TOP_UNIT
```

Then see https://ghdl.github.io/ghdl/using/Synthesis.html and the examples in this repo.

Welcome!

PPlinux, 2021-03-30:
Thanks,
yosys and eda installed.
AX8.vhd contains:
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use work.AX_Pack.all;

u0051702@N-NB107177 MINGW64 ~
$ ls
A90S1200.vhd  AX_ALU.vhd    AX_Pack.vhd  AX_Reg.vhd   AX_TC8.vhd       test.txt
A90S2313.vhd  AX_DPRAM.vhd  AX_Port.vhd  AX_Reg2.vhd  AX_UART.vhd      work
AX8.vhd       AX_PCS.vhd    AX_RAM.vhd   AX_TC16.vhd  project.project

u0051702@N-NB107177 MINGW64 ~
$ ghdl syntax AX8.vhd
AX8.vhd:57:10: unit "ax_pack" not found in library "work"
AX8.vhd:85:21: entity 'ax8' was not analysed

How can ghdl know where to find work.AX_Pack.all

Should I make a ghdl work directory for the A90S1200 (top) design ?

Greetings,

Patrick

umarcor, 2021-03-30:
It seems that all the sources are in subdir `trunk/rtl/vhdl`. Therefore, you should be able to execute `ghdl synth *.vhd -e TOP_UNIT_NAME` in that subdir.

umarcor, 2021-03-30:
> Should I make a ghdl work directory for the A90S1200 (top) design ?

I overlooked this. If you want to synthesize that specific design, try the following:

```
ghdl synth *.vhd -e A90S1200 > synthesized_A90S1200.vhd
```

PPlinux, 2021-03-30:
OK seems to do somthings.
Generates a long

umarcor, 2021-03-30:
> How to progress and simulate

For simulating, you need a testbench. In fact, the same testbench that works for pre-synthesis VHDL sources should work with the synthesized output too.

> generate FPGA bit file for Lattice ?
> Should be done with yosys I suppose ?

Yes, instead of using `ghdl --synth`, you need to call yosys. You need to also know which Lattice device you want to synthesize for. Then, after synthesis, you will need a PnR tool. I assume that you want to use an ICE40 or ECP5 device, so you'll need nextpnr (also available on MSYS2). Apart from the tools and VHDL sources, you need a constraint file, for telling which pins to connect to top-level ports.

- Find an example makefile for a blink design in https://github.com/im-tomu/fomu-workshop/tree/master/vhdl/blink. That is based on the [examples](https://github.com/ghdl/ghdl-yosys-plugin/tree/master/examples) in this repo, and also in https://github.com/antonblanchard/ghdl-yosys-blink.
- There are constraint files available in [hdl/constraints](https://github.com/hdl/constraints), which you might reuse or copy/modify.

PPlinux, 2021-03-30:
Thanks,
Will try (blink) firts tomorrow.
Greetings,
Patrick

closed 2021-10-10 by tgingold