ghdl/ghdl-yosys-plugin #148: GHDL synthesis - beginner question

etnapoli, 2021-04-08:
Working on windows 10. 
Installed (copied the files) for GHDL (ghdl-0.37-mingw32-mcode.zip)
This is the version:

GHDL 0.37 (v0.37) [Dunoon edition]
Compiled with GNAT Version: 9.1.0
 mcode code generator
Written by Tristan Gingold.

I read that synthesis should be available for 0.37 version. However if I type :
ghdl --help   
there is no synth option.
If I run :
ghdl --synth VHDL_file_name.VHD' 
I get the error:
"C:\cad\ghdl\bin\ghdl.exe: unknown command '--synth', try --help"
Same result with:
ghdl synth VHDL_file_name.VHD  

Any hint?
Thanks in advance

umarcor, 2021-04-08:
Hi @etnapoli!

In this case "since" does not include version 0.37. The commit that enabled it by default is the first commit after tag 0.37.0.

Anyway, on Windows the recommended solution is to install MSYS2 and then `pacman -S mingw-w64-x86_64-ghdl-llvm` or `pacman -S mingw-w64-i686-ghdl-mcode`. Then, it will be automatically updated when you update the system (`pacman -Syu`). Alternatively, you can  download the nightly `*.zst` packages and install them with `pacman -U *.zst`. See https://hdl.github.io/MINGW-packages/#_usage.

etnapoli, 2021-04-12:
Hi @umarcor 
Thank you very much for your answer. 
I followed your suggestion and it seems to work now.
Fo future readers. 
I'm working on windows 10. 
1) I installed MSYS2 (http://repo.msys2.org/distrib/msys2-x86_64-latest.exe)
2) I installed the gcc ( 32 bit version) running in the Msys2 terminal the command : pacman -S --needed base-devel mingw-w64-x86_64-toolchain
3) I installed ghdl and yosis : pacman -Sy mingw-w64-i686-ghdl-mcode     and    pacman -Sy mingw-w64-i686-yosys

At the end of the process in the directory: c:\msys64\mingw32\bin\  i have a working ghdl.exe and yosys.exe executables that I call including the entire path.

In my case the yosis command : yosis -m ghdl -p 'ghdl eetcetera...' does not work.
I created a command file 'synthesize.ys' and I run yosis as:
c:\msys64\mingw32\bin\yosys -s synthesize.ys

Thank you again.

closed 2021-04-12 by etnapoli

umarcor, 2021-04-12:
Nice @etnapoli! Your comments are very relevant:

> At the end of the process in the directory: c:\msys64\mingw32\bin\ i have a working ghdl.exe and yosys.exe executables that I call including the entire path.

This is a nice feature. Even though using MSYS2 for installing GHDL and Yosys is recommended, you don't need to use them from MINGW32/MINGW64 terminals/shells. They can be executed from cmd or powershell as you said.

> In my case the yosis command : yosis -m ghdl -p 'ghdl eetcetera...' does not work.

This is because Yosys on Windows does not support loading modules dynamically yet. When using `-m ghdl`, it is trying to search for a shared lib, and it cannot do that on Windows. Therefore, the Yosys version you get from MSYS2 has GHDL built in, as explained in https://github.com/ghdl/ghdl-yosys-plugin#build-as-part-of-yosys-not-recommended. In fact, the "not recommended" build procedure is maintained for this use case. On the one hand, this is why installing through MSYS2 is recommended (you avoid this building details/challenges). On the other hand, I submitted a patch to Yosys, so that the error message should explain it:

```
# yosys -m ghdl
...
ERROR: 
  This version of Yosys cannot load plugins at runtime.
  Some plugins may have been included at build time.
  Use option `-H' to see the available built-in and plugin commands.
  
# yosys -H | grep ghdl
  ghdl                 load VHDL designs using GHDL
```

> Thank you again.

You are welcome!