ghdl/ghdl-yosys-plugin #150: Build library type mismatch

jeinstei, 2021-06-10:
On a recent clone off master and a build, we were finding that there was a mismatch in the build process between the first system library found (libgnat.a) and the need for the shared object linking. To fix this, we had to add `-l:libgnat.so` to force the shared object library to have precedence. Not yet sure about a fix, but adding the explicit library call would at least solve issues on systems with multiple matching libraries. Haven't looked more in to it yet.

```
# Get sources
cd $FPGA_TOOLS
git clone https://github.com/ghdl/ghdl-yosys-plugin.git
cd ghdl-yosys-plugin
make clean
git pull

make
```

```
[ 10%] Building libyosys.so
clang version 10.0.0-4ubuntu1 
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/9
Found candidate GCC installation: /usr/lib/gcc/x86_64-linux-gnu/9
Selected GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/9
Candidate multilib: .;@m64
Candidate multilib: 32;@m32
Candidate multilib: x32;@mx32
Selected multilib: .;@m64
clang: warning: argument unused during compilation: '-pg' [-Wunused-command-line-argument]
 "/usr/bin/ld" -export-dynamic -z relro --hash-style=gnu --build-id --eh-frame-hdr -m elf_x86_64 -shared -export-dynamic -o libyosys.so /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/crti.o /usr/bin/../lib/gcc/x86_64-linux-gnu/9/crtbeginS.o -L/usr/bin/../lib/gcc/x86_64-linux-gnu/9 -L/usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu -L/lib/x86_64-linux-gnu -L/lib/../lib64 -L/usr/lib/x86_64-linux-gnu -L/usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../.. -L/usr/lib/llvm-10/bin/../lib -L/lib -L/usr/lib -soname /usr/local/lib/yosys/libyosys.so kernel/version_1667ad65.o kernel/python_wrappers.o kernel/register.o kernel/rtlil.o kernel/log.o kernel/calc.o kernel/yosys.o kernel/cellaigs.o kernel/celledges.o kernel/satgen.o kernel/mem.o kernel/ffmerge.o libs/bigint/BigIntegerAlgorithms.o libs/bigint/BigInteger.o libs/bigint/BigIntegerUtils.o libs/bigint/BigUnsigned.o libs/bigint/BigUnsignedInABase.o libs/sha1/sha1.o libs/json11/json11.o libs/subcircuit/subcircuit.o libs/ezsat/ezsat.o libs/ezsat/ezminisat.o libs/minisat/Options.o libs/minisat/SimpSolver.o libs/minisat/Solver.o libs/minisat/System.o frontends/ghdl/ghdl.o frontends/aiger/aigerparse.o frontends/rpc/rpc_frontend.o frontends/verific/verific.o frontends/json/jsonparse.o frontends/verilog/verilog_parser.tab.o frontends/verilog/verilog_lexer.o frontends/verilog/preproc.o frontends/verilog/verilog_frontend.o frontends/verilog/const2ast.o frontends/ast/ast.o frontends/ast/simplify.o frontends/ast/genrtlil.o frontends/ast/dpicall.o frontends/blif/blifparse.o frontends/liberty/liberty.o frontends/rtlil/rtlil_parser.tab.o frontends/rtlil/rtlil_lexer.o frontends/rtlil/rtlil_frontend.o passes/tests/test_autotb.o passes/tests/test_cell.o passes/tests/test_abcloop.o passes/sat/sat.o passes/sat/freduce.o passes/sat/eval.o passes/sat/sim.o passes/sat/miter.o passes/sat/expose.o passes/sat/assertpmux.o passes/sat/clk2fflogic.o passes/sat/async2sync.o passes/sat/supercover.o passes/sat/fmcombine.o passes/sat/mutate.o passes/sat/cutpoint.o passes/sat/fminit.o passes/sat/qbfsat.o passes/cmds/exec.o passes/cmds/add.o passes/cmds/delete.o passes/cmds/design.o passes/cmds/select.o passes/cmds/show.o passes/cmds/rename.o passes/cmds/autoname.o passes/cmds/connect.o passes/cmds/scatter.o passes/cmds/setundef.o passes/cmds/splitnets.o passes/cmds/stat.o passes/cmds/setattr.o passes/cmds/copy.o passes/cmds/splice.o passes/cmds/scc.o passes/cmds/torder.o passes/cmds/logcmd.o passes/cmds/tee.o passes/cmds/write_file.o passes/cmds/connwrappers.o passes/cmds/cover.o passes/cmds/trace.o passes/cmds/plugin.o passes/cmds/check.o passes/cmds/qwp.o passes/cmds/edgetypes.o passes/cmds/portlist.o passes/cmds/chformal.o passes/cmds/chtype.o passes/cmds/blackbox.o passes/cmds/ltp.o passes/cmds/bugpoint.o passes/cmds/scratchpad.o passes/cmds/logger.o passes/cmds/printattrs.o passes/memory/memory.o passes/memory/memory_dff.o passes/memory/memory_share.o passes/memory/memory_collect.o passes/memory/memory_unpack.o passes/memory/memory_bram.o passes/memory/memory_map.o passes/memory/memory_memx.o passes/memory/memory_nordff.o passes/memory/memory_narrow.o passes/techmap/flatten.o passes/techmap/techmap.o passes/techmap/simplemap.o passes/techmap/dfflibmap.o passes/techmap/maccmap.o passes/techmap/libparse.o passes/techmap/abc.o passes/techmap/abc9.o passes/techmap/abc9_exe.o passes/techmap/abc9_ops.o passes/techmap/iopadmap.o passes/techmap/clkbufmap.o passes/techmap/hilomap.o passes/techmap/extract.o passes/techmap/extract_fa.o passes/techmap/extract_counter.o passes/techmap/extract_reduce.o passes/techmap/alumacc.o passes/techmap/dffinit.o passes/techmap/pmuxtree.o passes/techmap/muxcover.o passes/techmap/aigmap.o passes/techmap/tribuf.o passes/techmap/lut2mux.o passes/techmap/nlutmap.o passes/techmap/shregmap.o passes/techmap/deminout.o passes/techmap/insbuf.o passes/techmap/attrmvcp.o passes/techmap/attrmap.o passes/techmap/zinit.o passes/techmap/dfflegalize.o passes/techmap/dffunmap.o passes/techmap/flowmap.o passes/techmap/extractinv.o passes/pmgen/test_pmgen.o passes/pmgen/ice40_dsp.o passes/pmgen/ice40_wrapcarry.o passes/pmgen/xilinx_dsp.o passes/pmgen/peepopt.o passes/pmgen/xilinx_srl.o passes/opt/opt.o passes/opt/opt_merge.o passes/opt/opt_mem.o passes/opt/opt_mem_feedback.o passes/opt/opt_muxtree.o passes/opt/opt_reduce.o passes/opt/opt_dff.o passes/opt/opt_share.o passes/opt/opt_clean.o passes/opt/opt_expr.o passes/opt/share.o passes/opt/wreduce.o passes/opt/opt_demorgan.o passes/opt/rmports.o passes/opt/opt_lut.o passes/opt/opt_lut_ins.o passes/opt/pmux2shiftx.o passes/opt/muxpack.o passes/hierarchy/hierarchy.o passes/hierarchy/uniquify.o passes/hierarchy/submod.o passes/equiv/equiv_make.o passes/equiv/equiv_miter.o passes/equiv/equiv_simple.o passes/equiv/equiv_status.o passes/equiv/equiv_add.o passes/equiv/equiv_remove.o passes/equiv/equiv_induct.o passes/equiv/equiv_struct.o passes/equiv/equiv_purge.o passes/equiv/equiv_mark.o passes/equiv/equiv_opt.o passes/fsm/fsm.o passes/fsm/fsm_detect.o passes/fsm/fsm_extract.o passes/fsm/fsm_opt.o passes/fsm/fsm_expand.o passes/fsm/fsm_recode.o passes/fsm/fsm_info.o passes/fsm/fsm_export.o passes/fsm/fsm_map.o passes/proc/proc.o passes/proc/proc_prune.o passes/proc/proc_clean.o passes/proc/proc_rmdead.o passes/proc/proc_init.o passes/proc/proc_arst.o passes/proc/proc_mux.o passes/proc/proc_dlatch.o passes/proc/proc_dff.o passes/proc/proc_memwr.o backends/firrtl/firrtl.o backends/simplec/simplec.o backends/edif/edif.o backends/cxxrtl/cxxrtl_backend.o backends/aiger/aiger.o backends/aiger/xaiger.o backends/protobuf/protobuf.o backends/protobuf/yosys.pb.o backends/table/table.o backends/btor/btor.o backends/intersynth/intersynth.o backends/json/json.o backends/verilog/verilog_backend.o backends/smv/smv.o backends/blif/blif.o backends/rtlil/rtlil_backend.o backends/smt2/smt2.o backends/spice/spice.o techlibs/easic/synth_easic.o techlibs/xilinx/synth_xilinx.o techlibs/xilinx/xilinx_dffopt.o techlibs/gowin/synth_gowin.o techlibs/ice40/synth_ice40.o techlibs/ice40/ice40_braminit.o techlibs/ice40/ice40_opt.o techlibs/nexus/synth_nexus.o techlibs/machxo2/synth_machxo2.o techlibs/intel/synth_intel.o techlibs/ecp5/synth_ecp5.o techlibs/ecp5/ecp5_gsr.o techlibs/coolrunner2/synth_coolrunner2.o techlibs/coolrunner2/coolrunner2_sop.o techlibs/coolrunner2/coolrunner2_fixup.o techlibs/efinix/synth_efinix.o techlibs/efinix/efinix_fixcarry.o techlibs/greenpak4/synth_greenpak4.o techlibs/greenpak4/greenpak4_dffinv.o techlibs/anlogic/synth_anlogic.o techlibs/anlogic/anlogic_eqn.o techlibs/anlogic/anlogic_fixcarry.o techlibs/achronix/synth_achronix.o techlibs/common/synth.o techlibs/common/prep.o techlibs/quicklogic/synth_quicklogic.o techlibs/intel_alm/synth_intel_alm.o techlibs/sf2/synth_sf2.o -lstdc++ -lm -lrt -lpython3.8 -lcrypt -lpthread -ldl -lutil -lm -lm -lboost_python38 -lboost_system -lboost_filesystem -lreadline -lffi -ldl -lz -ltcl8.6 -ltclstub8.6 /usr/local/lib/libghdl.a /usr/lib/gcc/x86_64-linux-gnu/9/adalib/libgnat.a -lprotobuf /usr/lib/llvm-10/lib/clang/10.0.0/lib/linux/libclang_rt.profile-x86_64.a -lstdc++ -lm -lgcc_s -lgcc -lpthread -lc -lgcc_s -lgcc /usr/bin/../lib/gcc/x86_64-linux-gnu/9/crtendS.o /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/crtn.o
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/adalib/libgnat.a(a-calend.o): relocation R_X86_64_PC32 against symbol `ada__calendar__leap_support' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:704: libyosys.so] Error 1
```

tgingold, 2021-06-10:
There is something I don't understand: why 'make' is building libyosys.so ?  The default make target is supposed to compile a plugin.

jeinstei, 2021-06-10:
That's a great question. I'm going to rerun things and see if I get the same results.

jeinstei, 2021-06-10:
The initial run was on Ubuntu 20.10, I'm now on 21.04, and can't reproduce this, and there has been a version bump to GCC10 with the new Ubuntu. I'll keep looking in to this. Our current fix works, so I'm just going to call this a non-event and close it until I can reproduce.

jeinstei, 2021-06-11:
Okay. Here's what I have. If `yosys` is not built with the `-l:libgnat.so` flag on both our Ubutnu systems on both GCC versions, we get an ld error for relocations as below. This is a problem with how libyosys is interacting with the ghdl-yosys-plugin, and I'm happy to look in to it further, but my best guess is that libyosys is being compiled with a bad reference to its gnat components.

```
$ make
yosys-config --build ghdl.so ghdl.o -shared /usr/local/lib/libghdl-2_0_0_dev.so -Wl,-rpath,/usr/local/lib/
clang: warning: argument unused during compilation: '-pg' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-pg' [-Wunused-command-line-argument]
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/10/adalib/libgnat.a(exit.o): warning: relocation against `gnat_exit_status' in read-only section `.text'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/10/adalib/libgnat.a(a-calend.o): relocation R_X86_64_PC32 against symbol `ada__calendar__leap_support' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:21: ghdl.so] Error 1
```

Our yosys build process is below, where you can see the plugin source getting copied in to the frontends folder.

```
# Setup GHDL
$DIR/setup_ghdl_yosys_plugin
cp -r $FPGA_TOOLS/ghdl-yosys-plugin/src frontends/ghdl
export GHDL_PREFIX=/usr/local
export LDFLAGS="$LDFLAGS -l:libgnat.so"

# Patch the Makefile
declare -a OPTIONS=("ENABLE_GHDL" "ENABLE_LIBYOSYS" "ENABLE_PROTOBUF" "ENABLE_PYOSYS" "ENABLE_GCOV" "ENABLE_GPROF" >
for OPTION in ${OPTIONS[@]}; do
  sed_string="s/$OPTION := 0/$OPTION := 1/"
  sed -i "$sed_string" Makefile
done

# Compile
make config-clang
make -j
sudo make install
make manual
```
After finishing the build with the explicit library call `-l:libgnat.so` added back in to the LDFLAGS, we can then standalone build the plugin:

```
$ make
yosys-config --build ghdl.so ghdl.o -shared /usr/local/lib/libghdl-2_0_0_dev.so -Wl,-rpath,/usr/local/lib/
clang: warning: argument unused during compilation: '-pg' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-pg' [-Wunused-command-line-argument]
```

tgingold, 2021-06-11:
When you build yosys, you must not add the ghdl plugin.

Xiretza, 2021-06-11:
Is there a good reason why you're using the "not recommended" approach to building the plugin instead of https://github.com/ghdl/ghdl-yosys-plugin#build-as-a-module-shared-library?

jeinstei, 2021-06-11:
> Is there a good reason why you're using the "not recommended" approach to building the plugin instead of https://github.com/ghdl/ghdl-yosys-plugin#build-as-a-module-shared-library?

I can't really give a good reason as to why we are using the not recommended approach. But we definitely should be switching over to doing it in the recommended manner. We'll try that out without the build modifications and see how that works out.

jeinstei, 2021-06-11:
I think what's you're actually seeing is a sort of hybrid where we are copying it in to yosys and building it, but ALSO running make standalone in the folder. The standalone make and yosys_config is what was causing the library errors above. We'll skip that copy step to src/frontends and see if it avoids this error.

jeinstei, 2021-06-11:
This happens when building standalone following the recommended instructions. Unless `LDFLAGS=$LDFLAGS -l:libgnat.so` is added to the make process for the plugin, we get the error as above about the relocation. Regardless of the flags used in building yosys.

oliverbm67, 2021-12-21:
I have the same issue with everything building from source :
- yosys (hash 7608985d2c6237b869a4774c6b1659282e7473ad)
- ghdl (hash 0def1b4e3223aead923308975450d400a52de2fe)
- plugin ( hash 09a32cd13c3335e3231d6ccd32c033826c6a3798)

Adding LDFLAGS=$LDFLAGS -l:libgnat.so do NOT solve the issue in my case. The make command output the following :
```
yosys-config --exec --cxx -c --cxxflags -o ghdl.o src/ghdl.cc -fPIC -DYOSYS_ENABLE_GHDL -I/usr/local/include -O -DGHDL_VER_HASH="\"09a32cd\""
yosys-config --build ghdl.so ghdl.o -shared /usr/local/lib/libghdl-2_0_0_dev.so -Wl,-rpath,/usr/local/lib/ -l:libgnat.so
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/10/adalib/libgnat.a(exit.o): warning: relocation against `gnat_exit_status' in read-only section `.text'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/10/adalib/libgnat.a(a-calend.o): relocation R_X86_64_PC32 against symbol `ada__calendar__leap_support' can not be used when making a shared object; recompilé avec -fPIC
/usr/bin/ld : échec de l'édition de liens finale : bad value
collect2: error: ld returned 1 exit status
make: *** [Makefile:23 : ghdl.so] Erreur 1
```
I am on Pop! OS 21.04 (Ubuntu 21.04 derivative), with gcc 10.3.0

jeinstei, 2021-12-21:
Does libgnat.so exist on your system?

umarcor, 2021-12-21:
@oliverbm67 did you build GHDL from sources?

oliverbm67, 2021-12-21:
I have the following when looking for libgnat.so
```
/usr/lib/x86_64-linux-gnu/libgnat.so
/usr/lib/gcc/x86_64-linux-gnu/10/adalib/libgnat.so
```
For GHDL, I first tried the package from the Ubuntu repo, and then remove it and built GHDL from sources

umarcor, 2021-12-21:
@oliverbm67 when building GHDL from sources, use configure argument `--default-pic`: https://github.com/ghdl/ghdl/blob/master/configure.

jeinstei, 2021-12-21:
@umarcor isn't that flag on by default? I need to try rebuilding from source in my new environment now, though.

umarcor, 2021-12-21:
@jeinstei it depends on the host. The configure script tries to detect if the toolchain on the host was compiled with PIC/fPIC. If so, it is enabled by default. However, sometimes it cannot detect it (or it is not enabled on the host). See https://github.com/ghdl/ghdl/blob/7e41be2dabf79b21f3d0be210e3d01d541a7e82c/configure#L309-L313

oliverbm67, 2021-12-21:
Looking into the GHDL Makefile, it seems to be enabled by default. Do you have a way to check if it was compiled correctly ?

oliverbm67, 2021-12-21:
results from running testsuite.sh
```
 tests: sanity pyunit gna vests synth vpi vhpi
> args: 
GHDL is: /usr/local/bin/ghdl
GHDL 2.0.0-dev (1.0.0.r943.g0def1b4e) [Dunoon edition]
 Compiled with GNAT Version: 10.3.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2021 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
REF: master
HASH: 0def1b4e3223aead923308975450d400a52de2fe
...
...

[GHDL - test] sanity
sanity 000hello: ok
sanity 001hello87: ok
sanity 002hello2008: ok
sanity 004all08: ok
sanity 005examples: ok
sanity 006upf: ok
sanity tests are successful
[GHDL - test] pyunit
/usr/bin/python3: No module named pytest

```

umarcor, 2021-12-21:
That means both GHDL and `libghdl.so` are correct. Therefore, the issue might be related to the `libgnat` available on the system.

Are you familiar with containers? Can you reproduce it consistently on a `ubuntu:21.04` container?

oliverbm67, 2021-12-21:
I am not very familiar with docker and containers, but I can give it a try

umarcor, 2021-12-22:
It is always annoying to not know the actual reason, but at least it's working for you. Can you now get the synthesis output you wanted?

oliverbm67, 2021-12-22:
A quick look on the latest commit for yosys do not seem to indicate any change relating to Makefile build settings...so I have no idea what the issue was. I am able to get synthesis results so thanks for your time. @jeinstei do you still require the extra -libgnat.so option in order to build the plugin ? If not I think this issue can be closed ?

jeinstei, 2021-12-22:
Definitely. I've had no building issues recently.

closed 2021-12-22 by jeinstei