ghdl/ghdl-yosys-plugin #153: VHDL to Verilog conversion with yosys-ghdl-plugin -- how to perserve signal names

71GA, 2021-08-10:
I have this **VHDL** design in a file `spi_icebreaker.vhdl`:

    -- A:
    library ieee;
    use ieee.std_logic_1164.all;
    
    -- B:
    entity spi_icebreaker is
    	port(
    		pcb_oscilator: in std_logic;
    		pcb_led: out std_logic;
    		pcb_button: in std_logic
    	);
    end spi_icebreaker;
    
    
    ----------------------------------------------------------------------------------------------------
    
    
    -- C:
    architecture logic_001 of spi_icebreaker is
    
    	-- D:
    	signal sbclki: std_logic;
    	signal sbrwi: std_logic;
    	signal sbstbi: std_logic;
    	signal sbadri: std_logic_vector (7 downto 0);
    	signal sbdati: std_logic_vector (7 downto 0);
    	signal sbdato: std_logic_vector (7 downto 0);
    	signal sbacko: std_logic;
    	signal spiirq: std_logic;
    	signal spiwkup: std_logic;
    	signal mi: std_logic;
    	signal so: std_logic;
    	signal soe: std_logic;
    	signal si: std_logic;
    	signal mo: std_logic;
    	signal moe: std_logic;
    	signal scki: std_logic;
    	signal scko: std_logic;
    	signal sckoe: std_logic;
    	signal scsni: std_logic;
    	signal mcsno: std_logic_vector (3 downto 0);
    	signal mcsnoe: std_logic_vector (3 downto 0);
    
    	-- E:
    	component SB_SPI is
    		generic(
    			BUS_ADDR74: integer
    		);
    		port(
    			SBCLKI: in std_logic; -- System bus: System clock input
    			SBRWI: in std_logic; -- System bus: System read/write input
    			SBSTBI: in std_logic; -- System bus: Strobe signal
    			SBADRI0: in std_logic; -- System bus: control registers address
    			SBADRI1: in std_logic;
    			SBADRI2: in std_logic;
    			SBADRI3: in std_logic;
    			SBADRI4: in std_logic;
    			SBADRI5: in std_logic;
    			SBADRI6: in std_logic;
    			SBADRI7: in std_logic;
    			SBDATI0: in std_logic; -- System bus: System data input
    			SBDATI1: in std_logic;
    			SBDATI2: in std_logic;
    			SBDATI3: in std_logic;
    			SBDATI4: in std_logic;
    			SBDATI5: in std_logic;
    			SBDATI6: in std_logic;
    			SBDATI7: in std_logic;
[... 248 lines cut ...]
        .SBDATI7(1'hx),
        .SBDATO0(_00_),
        .SBDATO1(_11_),
        .SBDATO2(_19_),
        .SBDATO3(_20_),
        .SBDATO4(_21_),
        .SBDATO5(_22_),
        .SBDATO6(_23_),
        .SBDATO7(_24_),
        .SBRWI(1'hx),
        .SBSTBI(1'hx),
        .SCKI(1'hx),
        .SCKO(_07_),
        .SCKOE(_08_),
        .SCSNI(1'hx),
        .SI(1'hx),
        .SO(_03_),
        .SOE(_04_),
        .SPIIRQ(_01_),
        .SPIWKUP(_02_)
      );
      assign pcb_led = _18_;
    endmodule

It bothers me that signal names are completely different in the **Verilog** file compared to the **VHDL** file. For example this **VHDL** signal `mo` is renamed to `_05_`... Why? How can I keep the original names? Does it have anything to do with the warnings that I get when translating **VHDL** to **Verilog**?

bluesceada, 2022-06-10:
In any of the two conversions, is it possible to convert vhdl generics -> verilog parameters? i.e. preserve the generics and don't fill them with their defaults?

umarcor, 2022-06-10:
@bluesceada the output of `ghdl synth` or the one of the plugin is a post-elaboration result. Therefore, alll the parameters are resolved by the time the ouput is generated. See http://ghdl.github.io/ghdl/internals/index.html.