ghdl/ghdl-yosys-plugin #155: Case sensitivity issue

SiliconWizard, 2021-09-14:
VHDL is case-insensitive, and one can assume GHDL doesn't care about case whatsoever.

There seems to be an issue with this though with ghdl-yosys-plugin, and more generally with the synthesis part of GHDL. Verilog *is* case-sensitive. So Yosys itself is.

When does that matter? Essentially with blackbox modules. If you instantiate some blackbox module from within VHDL, case sensitivity will matter for the output Verilog.

In most cases, GHDL synth seems to generate identifiers in all lowercase for its output Verilog, so that is good as long as any blackbox module you might be instantiating is also defined in all lowercase.

Cases for which it can cause an issue: currently using GHDL+Yosys for the Lattice ECP5. When instantiating ECP5 primitives, GHDL synth will sometimes generate the primitive identifier in all lowercase, and sometimes in all uppercase. That confuses Yosys, as it seems not to recognize the primitives correctly if they are in all lowercase.

Depending on the design, the exact same instantiation can yield a primitive in all uppercase or all lowercase, which is an issue, at least with Yosys.

Example:

> SDRAM_Clock: ODDRX1F
> 		port map (SCLK => Clock, RST => Reset, D0 => '0', D1 => '1', Q => SDRAM_CLK);

can either yield:

> ODDRX1F sdram_clock #(
>     .GSR("01000101010011100100000101000010010011000100010101000100"))
>     (
>     .SCLK(clock),
>     .RST(reset),
>     .D0(n337_o),
>     .D1(n338_o),
>     .Q(sdram_clock_Q));

or:

> oddrx1f sdram_clock #(
>     .gsr("01000101010011100100000101000010010011000100010101000100"))
>     (
>     .sclk(clock),
>     .rst(reset),
>     .d0(n540_o),
>     .d1(n541_o),
>     .q(sdram_clock_q));

same VHDL code, in the same source file, just a different project with different source files apart from this one.

tgingold, 2021-09-14:
Well, this is not an issue with a simple answer.

The default convention is to use the case of components and ports as they appear in the sources for blackboxes.

I don't know if this is the best choice.  Maybe we can have an option to use uppercase (as uppercase looks to be the convention).

Any suggestion is welcome!

SiliconWizard, 2021-09-14:
This convention looks fine to me.

But there's still a problem here in GHDL. As I said, in both cases, the component in VHDL code is in all uppercase, but GHDL sometimes translates it to all uppercase (so as in the VHDL source), and sometimes to all lowercase in the output Verilog, from the exact same VHDL source file. The only difference is that the hierarchy with the other entities is different.

I reproduced this with a project using a number of VHDL files... it would be unconvenient to send you the exact project at the moment. I haven't gotten around to reproducing this issue with a much simpler project that I could share here.

If GHDL implements the convention you mentioned, then maybe the issue I encountered is that in the failing case, GHDL doesn't recognize the component as a blackbox? But I'm 100% sure the implementation is not provided, as in cases for which it works correctly.

I ran into this issue with other primitives as well. The component definitions were all from this file: https://github.com/ghdl/ghdl-yosys-plugin/blob/master/library/ecp5u/components.vhdl

SiliconWizard, 2021-09-14:
Having said this and with your input, I took a look back at the 'components.vhdl' file above... and noticed you have defined all primitives in lowercase, whereas I use uppercase in my source files (as Lattice defines them in uppercase).

From a VHDL point of view, this mismatch doesn't cause any issue as it's not case-sensitive. But this mix of cases is maybe what makes GHDL sometimes generate the output names in all uppercase, or in all lowercase. Maybe depending on the order in which it processes files, or something like that.

From what I gathered, Yosys/nextpnr expect those primitives in all uppercase in order to recognize them. So it would make sense, IMO, to define them in all uppercase in the 'components.vhdl' file. Now I admit I used this file without knowing if it was "officially" supported and tested or not. Tell me what you think.

SiliconWizard, 2021-09-14:
OK, just tested, and that was it!

I changed the component name from "oddrx1f" to "ODDRX1F" in the 'components.vhdl' file above. I also had to change all port names to all uppercase for Yosys to recognize the primitive properly. I could run Yosys and Nextpnr to completion with this modification.

So, all component and port names in the 'components.vhdl' file for Lattice ECP5 should be changed to all uppercase.

hackfin, 2021-09-14:
@SiliconWizard You're right, should be uppercase. This file actually got updated with the proper caps, but it seems I forgot to update/file a PR for this project. It's in an incomplete component library, found here: https://section5.ch/downloads/ghdllib-1.0-devel.tgz.

tgingold, 2021-09-15:
A pull-request for components.vhdl is then welcome!

umarcor, 2021-09-20:
I guess we can close this, since #156 was merged?

tgingold, 2021-10-02:
Yes, closing.

closed 2021-10-02 by tgingold