ghdl/ghdl-yosys-plugin #157: Error: Info: No candidate top level module

sowana, 2021-09-25:
I am completely new to this project, but have installed and made all the required tools and they appear to work ok.
I created a simple half_adder.vhd file

library ieee;
use ieee.std_logic_1164.all;

entity half_adder is
  port (a, b: in std_logic;
    s, c: out std_logic); 
  end half_adder;

architecture behavior of half_adder is 
  begin
    ha: process (a, b)
    begin
      if a = '1' then
        s <= not b;
        c <= b;
      else
        s <= b;
        c <= '0'; 
      end if;
    end process ha;

end behavior;
Firstly I analysed the code
ghdl -a half_adder.vhd
then synthesised it 
yosys -m ghdl -p 'synth_ice40 -json half_adder.json' half_adder.vhd
all of the above completed without error.
then tried place and route
nectpnr-ice40 --up5k --pcf up5k.pcf --package sg48 --json  half_adder.json -asc half_adder,asc
this produced the following 
Info: No candidate top level modules.
ERROR: Failed to autodetect top module, please specify using --top.
0 warnings, 1 error
I not sure how to proceed as module is a verilog keyword and my source file vhdl.

rjordans, 2021-09-25:
I'd try `--top half_adder`, these tools were originally developed around verilog so you'll find some references to that in the output indeed.  The top module roughly translates to the topmost entity in your design

eine, 2021-09-27:
See https://github.com/ghdl/ghdl-yosys-plugin/blob/master/examples/ghdlsynth.mk#L18-L23.

sowana, 2021-09-27:
Thanks gents, much appreciated and all sorted.

closed 2021-09-27 by sowana