ghdl/ghdl-yosys-plugin #158: raised STORAGE_ERROR : stack overflow or erroneous memory access

suarezvictor, 2021-09-30:
There's a memory error processing VHDL files, it's not clear if this is a GHDL issue or just a plugin issue. It was open at GHDL github here: https://github.com/YosysHQ/yosys/issues/3022

I duplicate it here (files to uncompress are attached). Commands to be run are:
```
$ cd top
$ ghdl -i `cat ../vhdl_files.txt`
$ ghdl -m top
$ yosys -m ghdl
yosys> ghdl top

```
Results:
```
1. Executing GHDL.
Importing module top.
Importing module pmod_jb_0clk_23f04728.
Importing module pmod_jc_0clk_23f04728.
Importing module vga_0clk_83e31706.
Importing module vga_red_debug_output_main_0clk_23f04728.
Importing module vga_green_debug_output_main_0clk_23f04728.
Importing module vga_blue_debug_output_main_0clk_23f04728.
Importing module vsync_debug_output_main_0clk_23f04728.
Importing module hsync_debug_output_main_0clk_23f04728.
Importing module app_0clk_ebdb5920.

raised STORAGE_ERROR : stack overflow or erroneous memory access
```
[gldl error.tar.gz](https://github.com/ghdl/ghdl-yosys-plugin/files/7261300/gldl.error.tar.gz)

umarcor, 2021-10-01:
@suarezvictor, what version of GHDL, Yosys and ghdl-yosys-plugin are you using?

The following command works for me on MSYS2 (MINGW64) with up to date GHDL and Yosys (ghdl-yosys-plugin is built-in):

```sh
# cd top
# yosys -p "ghdl $(cat ../vhdl_files.txt) -e top;"

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+4300 (git sha1 551ef85cd, x86_64-w64-mingw32-g++ 10.3.0 -march=x86-64 -mtune=generic -O2 -Os)     

-- Running command `ghdl ../c_structs_pkg.pkg.vhd ../clk_cross_entities.vhd ../clk_cross_t_pkg.pkg.vhd ../top/top_5681.vhd ../examples/arty/src/vga/test_pattern.c/vga_blue_DEBUG_OUTPUT_MAIN/vga_blue_DEBUG_OUTPUT_MAIN_0CLK_23f04728.vhd ../examples/arty/src/vga/vga_pmod.c/vga/vga_0CLK_83e31706.vhd ../examples/arty/src/vga/test_pattern.c/vsync_DEBUG_OUTPUT_MAIN/vsync_DEBUG_OUTPUT_MAIN_0CLK_23f04728.vhd ../examples/arty/src/vga/test_pattern.c/app/app_0CLK_ebdb5920.vhd ../examples/arty/src/vga/../pmod/pmod_jc.c/pmod_jc/pmod_jc_0CLK_23f04728.vhd ../examples/arty/src/vga/../pmod/pmod_jb.c/pmod_jb/pmod_jb_0CLK_23f04728.vhd ../examples/arty/src/vga/test_pattern.c/vga_red_DEBUG_OUTPUT_MAIN/vga_red_DEBUG_OUTPUT_MAIN_0CLK_23f04728.vhd ../examples/arty/src/vga/test_pattern.c/hsync_DEBUG_OUTPUT_MAIN/hsync_DEBUG_OUTPUT_MAIN_0CLK_23f04728.vhd ../examples/arty/src/vga/test_pattern.c/vga_green_DEBUG_OUTPUT_MAIN/vga_green_DEBUG_OUTPUT_MAIN_0CLK_23f04728.vhd ../built_in/BIN_OP_EQ_uint1_t_uint1_t_uint1_t/BIN_OP_EQ_uint1_t_uint1_t_uint1_t_0CLK_de264c78.vhd ../built_in/MUX_uint1_t_uint1_t_uint1_t_uint1_t/MUX_uint1_t_uint1_t_uint1_t_uint1_t_0CLK_de264c78.vhd ../built_in/CONST_SR_0_uint4_t_uint4_t/CONST_SR_0_uint4_t_uint4_t_0CLK_de264c78.vhd ../built_in/MUX_uint12_t_uint1_t_uint12_t_uint12_t/MUX_uint12_t_uint1_t_uint12_t_uint12_t_0CLK_de264c78.vhd ../built_in/BIN_OP_PLUS_uint13_t_uint12_t_uint1_t/BIN_OP_PLUS_uint13_t_uint12_t_uint1_t_0CLK_de264c78.vhd ../built_in/BIN_OP_AND_uint1_t_uint1_t_uint1_t/BIN_OP_AND_uint1_t_uint1_t_uint1_t_0CLK_de264c78.vhd ../built_in/BIN_OP_LT_uint1_t_uint12_t_uint13_t/BIN_OP_LT_uint1_t_uint12_t_uint13_t_0CLK_de264c78.vhd ../built_in/BIN_OP_LT_uint1_t_uint12_t_uint10_t/BIN_OP_LT_uint1_t_uint12_t_uint10_t_0CLK_de264c78.vhd ../built_in/BIN_OP_EQ_uint1_t_uint12_t_uint11_t/BIN_OP_EQ_uint1_t_uint12_t_uint11_t_0CLK_de264c78.vhd ../built_in/BIN_OP_MINUS_uint12_t_uint12_t_uint1_t/BIN_OP_MINUS_uint12_t_uint12_t_uint1_t_0CLK_de264c78.vhd ../examples/arty/src/vga/test_pattern.c/vsync/vsync_0CLK_23f04728.vhd ../built_in/BIN_OP_LT_uint1_t_uint12_t_uint11_t/BIN_OP_LT_uint1_t_uint12_t_uint11_t_0CLK_de264c78.vhd ../built_in/BIN_OP_EQ_uint1_t_uint12_t_uint12_t/BIN_OP_EQ_uint1_t_uint12_t_uint12_t_0CLK_de264c78.vhd ../built_in/CONST_SR_3_uint4_t_uint4_t/CONST_SR_3_uint4_t_uint4_t_0CLK_de264c78.vhd ../built_in/CONST_SR_1_uint4_t_uint4_t/CONST_SR_1_uint4_t_uint4_t_0CLK_de264c78.vhd ../built_in/BIN_OP_GTE_uint1_t_uint12_t_uint12_t/BIN_OP_GTE_uint1_t_uint12_t_uint12_t_0CLK_de264c78.vhd ../built_in/BIN_OP_EQ_uint1_t_uint12_t_uint2_t/BIN_OP_EQ_uint1_t_uint12_t_uint2_t_0CLK_de264c78.vhd ../built_in/BIN_OP_LT_uint1_t_uint12_t_uint9_t/BIN_OP_LT_uint1_t_uint12_t_uint9_t_0CLK_de264c78.vhd ../built_in/UNARY_OP_NOT_uint1_t_uint1_t/UNARY_OP_NOT_uint1_t_uint1_t_0CLK_de264c78.vhd ../built_in/MUX_uint4_t_uint1_t_uint4_t_uint4_t/MUX_uint4_t_uint1_t_uint4_t_uint4_t_0CLK_de264c78.vhd ../built_in/CONST_SR_2_uint4_t_uint4_t/CONST_SR_2_uint4_t_uint4_t_0CLK_de264c78.vhd ../built_in/BIN_OP_OR_uint1_t_uint1_t_uint1_t/BIN_OP_OR_uint1_t_uint1_t_uint1_t_0CLK_de264c78.vhd ../examples/arty/src/vga/test_pattern.c/vga_red/vga_red_0CLK_23f04728.vhd ../examples/arty/src/vga/test_pattern.c/vga_green/vga_green_0CLK_23f04728.vhd ../examples/arty/src/vga/test_pattern.c/hsync/hsync_0CLK_23f04728.vhd ../built_in/MUX_uint25_t_uint1_t_uint25_t_uint25_t/MUX_uint25_t_uint1_t_uint25_t_uint25_t_0CLK_de264c78.vhd ../examples/arty/src/vga/test_pattern.c/vga_blue/vga_blue_0CLK_23f04728.vhd ../built_in/BIN_OP_EQ_uint1_t_uint25_t_uint20_t/BIN_OP_EQ_uint1_t_uint25_t_uint20_t_0CLK_de264c78.vhd ../built_i [... cut ...]
Importing module pmod_jb_0clk_23f04728.
Importing module pmod_jc_0clk_23f04728.
Importing module vga_0clk_83e31706.
Importing module vga_red_debug_output_main_0clk_23f04728.
Importing module vga_green_debug_output_main_0clk_23f04728.
Importing module vga_blue_debug_output_main_0clk_23f04728.
Importing module vsync_debug_output_main_0clk_23f04728.
Importing module hsync_debug_output_main_0clk_23f04728.
Importing module app_0clk_ebdb5920.

raised STORAGE_ERROR : stack overflow or erroneous memory access
```

tgingold, 2021-10-02:
This is due to an infinite recursion.
In entity `app_0CLK_ebdb5920`, some nets of `read_pipe` are assigned in a circular way.

The design is clearly wrong, but a better error message is required.

tgingold, 2021-10-02:
But synthesizing `app_0CLK_ebdb5920` only works.  Weird.

JulianKemmerer, 2021-10-02:
@tgingold I can try to poke at the generated `app_0CLK` entity see if I can find the circularity you describe. Did you say you see this in the netlist directly?

There are lines like so in the vhdl I could try to remove
```vhdl
  -- Some tools dont like if read_pipe is never fully driven, dummy drive
  read_pipe := write_pipe;
 ```
 that look circular but arent really + lots of other variable usage that looks potentially circular shouldnt be.

This design (VGA test pattern gen) synthesizes and work in hardware fine (without and comb. loop warnings/errors)

Also - this memory error from ghdl/yosys doesnt occur all the time in my testing only certain runs, certain batches of generated code - where the only difference should be like those hex code in names ex. app_0clk `_ebdb5920`

Hmmm...

tgingold, 2021-10-02:
This one is circular: the process starts with:

    write_pipe := read_pipe;

and finishes with:

    read_pipe := write_pipe;

Some elements of this record are never assigned, so they are self-assigned.
And currently, this ends with an infinite recursion.

Removing the last line (read_pipe := write_pipe) fixes the issue.

JulianKemmerer, 2021-10-02:
Ah yes ok classic - "trying to support some tools 
```
  -- Some tools dont like if read_pipe is never fully driven, dummy drive
  read_pipe := write_pipe;
```
And ending up breaking others"

tgingold, 2021-10-02:
I plan to fix this issue!

JulianKemmerer, 2021-10-02:
A few notes I would want cleared up if possible @tgingold and thanks so much for planning have some fix on your end :call_me_hand: right on

```
This one is circular: the process starts with:

    write_pipe := read_pipe;

and finishes with:

    read_pipe := write_pipe;
```
This is what I meant with ~looking circular but not actually being. These are variables local to a process using blocking assignments. 
Seems like something like
```
write_pipe := read_pipe;
read_pipe := write_pipe;
```
should just evaluate to wires if I understand variables correctly - but I'd imagine it gets complicated with logic in between those expressions

JulianKemmerer, 2021-10-02:
Like there is no question the process loops back reevaluating itself as the signals propagate, sensitivity list, etc
And could absolutely form a loop - and in this case I was silly to form it like above I think.
Because in this case its a question of the record elements of `write_pipe` and `read_pipe` as you also mentioned - how some are never assigned

You mention `Some elements of this record are never assigned, so they are self-assigned.` and so I as you indicated I think you are onto some kind of fix in this regard eh.

Again thanks so much for your time

suarezvictor, 2021-10-02:
The version I'm using with the stack overflow issue is a Linux version tool compiled from github sources. Specifically:

```
$ ghdl --version
GHDL 2.0.0-dev (1.0.0.r794.g67ceca18) [Dunoon edition]
 Compiled with GNAT Version: 8.3.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2021 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
$ yosys --version
Yosys 0.9+4303 (git sha1 c88eaea6, clang 7.0.1-8+deb10u2 -fPIC -Os)
```

```
$ uname -a
Linux maguilaptop 3.16.0-4-amd64 #1 SMP Debian 3.16.51-3 (2017-12-13) x86_64 GNU/Linux
```

tgingold, 2021-10-02:
Should now be fixed.  Thank you for the issue.

closed 2021-10-02 by tgingold

JulianKemmerer, 2021-10-03:
Hello @tgingold I might be seeing a related issue.

`ERROR: Assert `n.id != 0' failed in frontends/ghdl/ghdl.cc:204.`

I was using verilator and seeing some circular logic warnings and thought about the before mentioned very circular looking 
```vhdl
  write_pipe := read_pipe;
  ...
  -- Some tools dont like if read_pipe is never fully driven, dummy drive
  read_pipe := write_pipe;
```
perhaps ending up with some kind of verilator circular loop too. So I commented it out
```vhdl
-- read_pipe := write_pipe;
```
Again I saw the GHDL warnings I was trying to avoid - but they are just warnings right?
example:
```
no assignment for offsets 0:171 of variable "read_pipe"
```

However, it seems after getting last nights oss cad build I now hit that `ghdl.cc:204` assertion.

Would you like a new github issue for this assertion hit?

Do you expect it related to 'unassigned variable record elements'? I.e. What does it mean to hit this?

Thanks again for your time.

tgingold, 2021-10-04:
Yes, please open a new issue as I was not able to reproduce this crash.

I don't know if this is related to the warning, which indeed is due to non-initialized elements in write_pipe.

commit a0b84cc5 (2021-10-02): ghdl.cc: always create a wire for signal/isignal.  Fix #158