ghdl/ghdl-yosys-plugin #161: make fails: unknown commands: "--libghdl-library-path" & "--libghdl-include-dir"

tortik92, 2021-11-14:
Hi!
I'm trying to make ghdl-yosis-plugin, but have some problems with make execution:
```
/usr/bin/ghdl-mcode:error: unknown command '--libghdl-library-path', try --help
/usr/bin/ghdl-mcode:error: unknown command '--libghdl-include-dir', try --help
yosys-config --exec --cxx -c --cxxflags -o ghdl.o src/ghdl.cc -fPIC -DYOSYS_ENABLE_GHDL -I -O -DGHDL_VER_HASH="\"9e11f71\""
/bin/sh: 1: yosys-config: not found
make: *** [Makefile:26: ghdl.o] Error 127
```
GHDL Version:
    GHDL 0.37
    GNAT: 9.3.0

Yosys Version:
    Yosys 0.9
    
Thank you for helping me!

tgingold, 2021-11-14:
I think you ghdl version is too old; you'd better to download either version 1.0, or rebuild it from sources.
In addition to that, you need `yosys-config`.  It is provided by yosys to build plugins.  So either you have to install a specific package or to rebuild yosys from sources.

tortik92, 2021-11-14:
Well, I installed it via "sudo apt-get ghdl" today,
yosys-dev was installed, and now I have some changes in error message:

```
/usr/bin/ghdl-mcode:error: unknown command '--libghdl-library-path', try --help
/usr/bin/ghdl-mcode:error: unknown command '--libghdl-include-dir', try --help
yosys-config --exec --cxx -c --cxxflags -o ghdl.o src/ghdl.cc -fPIC -DYOSYS_ENABLE_GHDL -I -O -DGHDL_VER_HASH="\"9e11f71\""
src/ghdl.cc:30:10: fatal error: ghdl/synth.h: No such file or directory
   30 | #include "ghdl/synth.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:26: ghdl.o] Error 1
```

tortik92, 2021-11-14:
Actually it is not closed

tortik92, 2021-11-14:
> 
> I think you ghdl version is too old; you'd better to download either version 1.0, or rebuild it from sources. In addition to that, you need `yosys-config`. It is provided by yosys to build plugins. So either you have to install a specific package or to rebuild yosys from sources.

I installed yosys-dev, and do have small changes in my error message:

```
/usr/bin/ghdl-mcode:error: unknown command '--libghdl-library-path', try --help
/usr/bin/ghdl-mcode:error: unknown command '--libghdl-include-dir', try --help
yosys-config --exec --cxx -c --cxxflags -o ghdl.o src/ghdl.cc -fPIC -DYOSYS_ENABLE_GHDL -I -O -DGHDL_VER_HASH="\"9e11f71\""
src/ghdl.cc:30:10: fatal error: ghdl/synth.h: No such file or directory
   30 | #include "ghdl/synth.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:26: ghdl.o] Error 1
```

tgingold, 2021-11-14:
So, the second problem is fixed: you now have yosys-config.
The first one is not.  The ghdl version provided by debian (or ubuntu or ??) is still too old.
You should recompile it from github.com/ghdl/ghdl
(or get the lastest snapshot from there)

Xiretza, 2021-11-14:
> I think you ghdl version is too old; you'd better to download either version 1.0, or rebuild it from sources.

tortik92, 2021-11-14:
Well, I admit, I am pretty bad at understnding of basic prinicples of linux.
I downloaded ghdl via "git clone https://github.com/ghdl/ghdl.git", but there is no makefile or something else to execute setup process of installation...

umarcor, 2021-11-14:
@tortik92 what is your operating system?

tortik92, 2021-11-14:
My OS is Windows 10 Home and I am running Ubuntu 20.04.3 LTS on WSL.

> 
> 
> @tortik92 what is your operating system?

umarcor, 2021-11-14:
You can either use MSYS2 (MINGW64, see https://www.msys2.org/) and install https://github.com/ghdl/ghdl/releases/download/nightly/mingw-w64-x86_64-ghdl-llvm-ci-1-any.pkg.tar.zst through `pacman -U *.zst`. Or extract https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-gha-ubuntu-20.04-mcode.tgz on WSL and add it to the PATH. That will avoid having to build GHDL from sources.

tortik92, 2021-11-14:
OK, I downloaded nightly version to my user folder. What exactly do I need to add to the PATH? bin folder?

umarcor, 2021-11-14:
Yes.

tortik92, 2021-11-14:
OK,
Is this
`export PATH="ghdl-gha-ubuntu-20.04-mcode/bin"`

right?

umarcor, 2021-11-14:
Typically, an absolute path is used. I don't know if a relative would work...

tortik92, 2021-11-14:
Absolute is then "/home/toritk92/ghdl-gha-ubuntu-20.04-mcode/bin"?

> 
> 
> Typically, an absolute path is used. I don't know if a relative one would work...

umarcor, 2021-11-14:
Assuming that you extracted it to the home directory of your user, and that your user name is `toritk92`, yes. However, note that typo: your username in GitHub is `tortik92`, not `toritk92`. Furthermore, `~/ghdl-gha-ubuntu-20.04-mcode/bin` should equivalent, since `~` is a shortcut for the home folder of the current user.

tortik92, 2021-11-14:
Ok, "toritk92" was just a type error in my message.

And... new error message!

```
Command 'make' is available in the following places
 * /bin/make
 * /usr/bin/make
The command could not be located because '/usr/bin:/bin' is not included in the PATH environment variable.
make: command not found
```

I was in "tortik92/ghdl-yosys-plugin"

umarcor, 2021-11-14:
By doing `export PATH="ghdl-gha-ubuntu-20.04-mcode/bin"`, you removed the previous content of the PATH, which contained some of those locations (`/bin`, `/usr/bin`). You should not replace PATH, but append to it. See https://unix.stackexchange.com/questions/26047/how-to-correctly-add-a-path-to-path

tortik92, 2021-11-14:
Hmm, Ok, I was wundering if it is actually creating a list of PATHs somewhere...
I remade my PATHs, so now I have as output of echo $PATH:
`~/ghdl-gha-ubuntu-20.04-mcode/bin:/usr/lib/lightdm/lightdm:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games`

after make:
```
/usr/bin/ghdl-mcode:error: unknown command '--libghdl-library-path', try --help
/usr/bin/ghdl-mcode:error: unknown command '--libghdl-include-dir', try --help
yosys-config --exec --cxx -c --cxxflags -o ghdl.o src/ghdl.cc -fPIC -DYOSYS_ENABLE_GHDL -I -O -DGHDL_VER_HASH="\"9e11f71\""
src/ghdl.cc:30:10: fatal error: ghdl/synth.h: No such file or directory
   30 | #include "ghdl/synth.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:26: ghdl.o] Error 1
```
Next circle of hell...

umarcor, 2021-11-14:
For some reason, it is finding the global GHDL, and not your local one. Note `/usr/bin/ghdl-mcode:`. You can check `which ghdl`.
Maybe you can try executing the local GHDL explicitly first: `~/ghdl-gha-ubuntu-20.04-mcode/bin/ghdl version`.

tortik92, 2021-11-14:
```
root@LAPTOP-TQ37UJ6D:/home/tortik92/ghdl-gha-ubuntu-20.04-mcode/bin# ghdl -v
GHDL 0.37 (Ubuntu 0.37+dfsg-1ubuntu1) [Dunoon edition]
 Compiled with GNAT Version: 9.3.0
 mcode code generator
Written by Tristan Gingold.
```

> 
> 
> For some reason, it is finding the global GHDL, and not your local one. Note `/usr/bin/ghdl-mcode:`. You can check `which ghdl`. Maybe you can try executing the local GHDL explicitly first: `~/ghdl-gha-ubuntu-20.04-mcode/bin/ghdl version`.

umarcor, 2021-11-14:
That's not what I suggested. Even though you are located in that subdir, by executing `ghdl` it is picking the global one, thus the one in `/usr/bin`. That's seen because of the version: 0.37. You need to execute the local binary.

tortik92, 2021-11-14:
OK, how do I do it?

umarcor, 2021-11-14:
https://stackoverflow.com/questions/9477157/how-to-run-binary-file-in-linux

tortik92, 2021-11-14:
> 
> 
> https://stackoverflow.com/questions/9477157/how-to-run-binary-file-in-linux

Thank you, today I will learn much of linux.

So... output is:

```
GHDL 2.0.0-dev (v1.0.0-870-ge3c865e8) [Dunoon edition]
 Compiled with GNAT Version: 9.3.0
 mcode code generator
Written by Tristan Gingold.
```

As long as I understand, when I do "make", it is referring to the old "ghdl", which is global and is outdated, and now, I need to change it, to the newer one...

umarcor, 2021-11-14:
That is nice! So, there is some problem with the PATH. You need to fix that. But, otherwise, the "new" version of GHDL seems to be properly executed. Maybe you can:

```sh
mv ~ghdl-gha-ubuntu-20.04-mcode /opt/ghdl
export PATH=/opt/ghdl/bin:"$PATH"
echo "$PATH"
which ghdl
ghdl version
```

tortik92, 2021-11-14:
`mv ~/ghdl-gha-ubuntu-20.04-mcode /opt/ghdl`
For some reason, ~ is "root" and not "home/tortik92"
`mv: cannot stat '/root/ghdl-gha-ubuntu-20.04-mcode': No such file or directory`

umarcor, 2021-11-14:
I suggest you close the WSL terminal and you open it again. You probably changed the user at some point through `su`. Plus, in a new terminal you get a clean PATH.

tortik92, 2021-11-14:
Thank you!
My PATH Output is:
`/opt/ghdl/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/lib/wsl/lib:/mnt/c/Program Files/WindowsApps/CanonicalGroupLimited.UbuntuonWindows_2004.2020.424.0_x64__79rhkp1fndgsc:/mnt/c....`

which ghdl:
```
tortik92@LAPTOP-TQ37UJ6D:~$ which ghdl
/opt/ghdl/bin/ghdl
```

GHDL Version:
```
tortik92@LAPTOP-TQ37UJ6D:~$ ghdl version
GHDL 2.0.0-dev (v1.0.0-870-ge3c865e8) [Dunoon edition]
 Compiled with GNAT Version: 9.3.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2021 Tristan Gingold.
...
```

tortik92, 2021-11-14:
After make (again):
A new couple of errors:

```
tortik92@LAPTOP-TQ37UJ6D:~/ghdl-yosys-plugin$ make
yosys-config --exec --cxx -c --cxxflags -o ghdl.o src/ghdl.cc -fPIC -DYOSYS_ENABLE_GHDL -I/opt/ghdl/include -O -DGHDL_VER_HASH="\"9e11f71\""
src/ghdl.cc: In function ‘void import_module(Yosys::RTLIL::Design*, GhdlSynth::Module)’:
src/ghdl.cc:637:30: error: ‘ID’ has not been declared
  637 |   module->set_bool_attribute(ID::blackbox);
      |                              ^~
src/ghdl.cc:657:59: error: no match for call to ‘(Yosys::hashlib::pool<Yosys::RTLIL::IdString>) (std::string)’
  657 |    module->avail_parameters(to_str(get_param_name(m, idx)));
      |                                                           ^
src/ghdl.cc:976:12: error: ‘struct Yosys::RTLIL::Module’ has no member named ‘addModFloor’
  976 |    module->addModFloor(to_str(iname), IN(0), IN(1), OUT(0), id == Id_Smod);
      |            ^~~~~~~~~~~
In file included from src/ghdl.cc:19:
src/ghdl.cc: At global scope:
/usr/share/yosys/include/kernel/yosys.h:130:34: fatal error: opening dependency file ghdl.d: Permission denied
  130 | #define YOSYS_NAMESPACE_END      }
      |                                  ^
src/ghdl.cc:1295:1: note: in expansion of macro ‘YOSYS_NAMESPACE_END’
 1295 | YOSYS_NAMESPACE_END
      | ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:26: ghdl.o] Error 1
```

umarcor, 2021-11-14:
I fear that might be because of an outdated version of Yosys. Did you install it through `apt`? Which version of Yosys is it?

tortik92, 2021-11-14:
Installed today to, probably via apt, but no more sure...

```
tortik92@LAPTOP-TQ37UJ6D:~/ghdl-yosys-plugin$ yosys -V
Yosys 0.9 (git sha1 1979e0b)
```

tortik92, 2021-11-14:
If you are interested in, where does this crusade go:
https://github.com/cpldcpu/PCBFlow

umarcor, 2021-11-14:
Unfortunately, that is very outdated. It's from August 2019: https://github.com/YosysHQ/yosys/commit/1979e0b. It seems they don't provide an standalone zipfile/tarball which you can extract. Hence, you will need to build Yosys from sources: https://github.com/YosysHQ/yosys#setup. Alternatively, the bundle they provide in https://github.com/YosysHQ/oss-cad-suite-build does include both GHDL and Yosys. See also [hdl/packages](https://github.com/hdl/packages).

tortik92, 2021-11-14:
Well, I will try second variant.

umarcor, 2021-11-14:
Nice. I'm closing this because the issue itself is solved.

closed 2021-11-14 by umarcor

tortik92, 2021-11-14:
Well, oss-cad-suite explanation of installation is a little bit tricky, but as long as I understand, it is now installing something...

tortik92, 2021-11-14:
[image]

tortik92, 2021-11-14:
But if poss-cad include GHDL & yosys, where is yosys in this list of loading elements?

umarcor, 2021-11-14:
Third one starting from the bottom.

tortik92, 2021-11-14:
Ohh... thanks=)
This is beeing a torture for... 7 hours for me...
But the result costs it

tortik92, 2021-11-14:
```
==> Step [ 1/50] building ghdl ...
  -> Remove old output dir ...
  -> Creating output dir ...
  -> Remove old build dir ...
  -> Creating build dir ...
  -> Copy 'ghdl' source to build dir ...
  -> Compiling ...
Traceback (most recent call last):
  File "./builder.py", line 85, in <module>
    cli(None)
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "./builder.py", line 38, in build
    buildCode(target, arch, nproc, force, dry, src, single, tar)
  File "/home/tortik92/oss-cad-suite-build/src/base.py", line 628, in buildCode
    code = executeBuild(target, arch, prefix, build_dir if not target.top_package else output_dir, output_dir, nproc, pack_sources)
  File "/home/tortik92/oss-cad-suite-build/src/base.py", line 431, in executeBuild
    return run_live(params, cwd=build_dir)
  File "/home/tortik92/oss-cad-suite-build/src/base.py", line 339, in run_live
    return asyncio.get_event_loop().run_until_complete(run_process(command, cwd, env))
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/tortik92/oss-cad-suite-build/src/base.py", line 318, in run_process
    process = await asyncio.create_subprocess_exec(*command, cwd=cwd, env=env,
  File "/usr/lib/python3.8/asyncio/subprocess.py", line 236, in create_subprocess_exec
    transport, protocol = await loop.subprocess_exec(
  File "/usr/lib/python3.8/asyncio/base_events.py", line 1630, in subprocess_exec
    transport = await self._make_subprocess_transport(
  File "/usr/lib/python3.8/asyncio/unix_events.py", line 197, in _make_subprocess_transport
    transp = _UnixSubprocessTransport(self, protocol, args, shell,
  File "/usr/lib/python3.8/asyncio/base_subprocess.py", line 36, in __init__
    self._start(args=args, shell=shell, stdin=stdin, stdout=stdout,
  File "/usr/lib/python3.8/asyncio/unix_events.py", line 789, in _start
    self._proc = subprocess.Popen(
  File "/usr/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'docker'
```

Well, directory "docker" exists...

tortik92, 2021-11-14:
I think, I will go to the oss-cad-suite-build site and open there a new issue... But then, I will return here again.