ghdl/ghdl-yosys-plugin #162: Upload of ghdl-yosys-plugin to Debian?

smoe, 2021-12-06:
Hello,
both ghdl and yosys are in Debian, but to the best of my understanding this connective plugin is not. Would this be supported from your side? While I could help with an initial upload, would there be a maintainer of the package amongst yours? I am aware of https://salsa.debian.org/electronics-team/ghdl which looks like that place is waiting for it.
Many thanks!
Steffen

tgingold, 2021-12-06:
I do maintain ghdl-yosys-plugin upstream (the original repo), but I don't want to maintain a distribution package.
However I suppose this is not a lot of work.

umarcor, 2021-12-07:
Although we don't have fluent communication with Debian packagers (the electronics team at salsa), they are doing a good job providing all the GHDL backends and allowing users to switch between them. Adding ghdl-yosys-plugin to their recipes should not increase the maintenance burden significantly.

A couple of months ago @andreasbombe stepped by (ghdl/ghdl#1885). Let me drop some references here, and let's hope that he replies to this issue:

- On Linux, ghdl-yosys-plugin can be built as a shared library. That is the recommended approach.
- The shared library requires GHDL and Yosys in order to be built.
- The shared library can be installed in Yosys' default plugins location.
- Yosys is a runtime dependency of the plugin. The GHDL binary is not a runtime dependency, because the plugin uses libghdl. However, both GHDL and libghdl do need the VHDL sources (std and ieee libraries) that are installed along with GHDL. Hence, in practice, the plugin depends on GHDL.

Assuming that both GHDL and Yosys are available on the system, this is how I recommend to build and install the plugin:

```
git clone https://github.com/ghdl/ghdl-yosys-plugin
cd ghdl-yosys-plugin
make
cp ghdl.so /usr/local/lib/ghdl_yosys.so

yosys-config --exec mkdir -p --datdir/plugins
yosys-config --exec ln -s /usr/local/lib/ghdl_yosys.so --datdir/plugins/ghdl.so
```

See https://github.com/ghdl/ghdl-yosys-plugin/blob/master/ci.sh#L68-L77 or https://github.com/hdl/containers/blob/main/debian-bullseye/ghdl-yosys-plugin.dockerfile#L41-L47.

DanielG, 2022-01-11:
FYI I'm working on packaging ghdl-yosys-plugin for Debian, see https://salsa.debian.org/dxld-guest/yosys-plugin-ghdl

Merging of #166 would be appreciated to reduce the patch count.

--Daniel

umarcor, 2022-01-11:
> FYI I'm working on packaging ghdl-yosys-plugin for Debian, see https://salsa.debian.org/dxld-guest/yosys-plugin-ghdl

I cannot access. Maybe it's a private repo?

> Merging of #166 would be appreciated to reduce the patch count.

Done!

DanielG, 2023-06-14:
It's in unstable/trixie now: https://tracker.debian.org/pkg/yosys-plugin-ghdl. So this can be closed.

closed 2023-06-15 by smoe

umarcor, 2023-06-22:
That's great news! Thanks for your contributions for Debian and for keeping us up to date! :heart: