ghdl/ghdl-yosys-plugin #163: wire not found for $posedge

kammoh, 2021-12-31:
Hi,
I'm not sure if this is a duplicate of #110, #128 (supposedly fixed?) or #146, but I'm pretty sure this code (which implements a pipelined SIPO) should be perfectly valid synthesizable VHDL and works fine with Vivado and all other tools I've tried. (Well, this cut-down MWE was not fully verified, but I'm confident about the original code).

```VHDL
library ieee;
  use ieee.std_logic_1164.all;

package mwe_pkg is
  type slv_array is array(natural range <>) of std_logic_vector(31 downto 0);
end package mwe_pkg;

library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

library work;
  use work.mwe_pkg.all;

entity mwe is
  port (
    clk       : in    std_logic;
    reset     : in    std_logic;
    in_data   : in    std_logic_vector(31 downto 0);
    in_valid  : in    std_logic;
    in_ready  : out   std_logic;
    out_data  : out   slv_array(0 to 15);
    out_valid : out   std_logic;
    out_ready : in    std_logic
  );
end entity mwe;

architecture rtl of mwe is

  signal mem  : slv_array(0 to 15);
  signal ctr  : unsigned(3 downto 0);
  signal full : std_logic;

begin

  out_valid <= full;
  in_ready  <= not full or out_ready;

  process (clk) is
  begin

    if rising_edge(clk) then
      if (reset = '1') then
        ctr  <= (others => '0');
        full <= '0';
      else
        if ((in_valid and in_ready) = '1') then
          mem(to_integer(ctr)) <= in_data;
          ctr                  <= ctr + 1;
          if (ctr = 15) then
            full <= '1';
          end if;
        end if;
        if ((out_valid and out_ready) = '1') then
          full <= '0';
        end if;
      end if;
    end if;

  end process;

  out_data <= mem;
end architecture rtl;
```

```
$ yosys -m ghdl -p "ghdl --std=08 synth_mwe.vhdl -e"
Yosys 0.12+45 (git sha1 cfe940a98, x86_64-apple-darwin20.2-clang 10.0.0-4ubuntu1 -fPIC -Os)
-- Running command `ghdl --std=08 synth_mwe.vhdl -e' --

1. Executing GHDL.
note: top entity is "mwe"
Importing module mwe.
ERROR: wire not found for $posedge
```

```
$ ghdl --version
GHDL 2.0.0-dev (1.0.0.r950.g8d512a44) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 mcode code generator
```

tgingold, 2022-01-03:
Hello,

I can reproduce the issue, but it is due to 'mem', which is only written.  Do you have a more complete reproducer ?
I could try to fix this one but I fear the final one may behave differently.

kammoh, 2022-01-06:
Hi @tgingold,
Thanks for looking into it.

> I can reproduce the issue, but it is due to 'mem', which is only written. Do you have a more complete reproducer ? I could try to fix this one but I fear the final one may behave differently.

I'm not sure I understand your comment. `mem` signal is both written and read (the line before last in the MWE):
```vhdl
  out_data <= mem;
```
The MWE should be implementing an actual SIPO (assuming I haven't introduced any bugs).

tgingold, 2022-01-07:
Yes, you're right.  I was focused on the process only.