ghdl/ghdl-yosys-plugin #165: Use of BRAM in ICE40 ...

PPlinux, 2022-01-08:
Dear,

can anyone help me using BRAM 512 word, 16 bit as a ROM (RAM preloaded during FPGA power on)
using GHDL VHDL -> Yosys -> Lattice ICE40 FPGA please ?

Greetings,

Patrick

kevingoh, 2022-07-21:
I am new to the toolchain and am having trouble understanding how the techmap works in the plugin. 

I have two entities, one is designed to be ROM while another as RAM. 
The ROM was mapped to BRAM while the RAM to DFFs.

I have no idea how to change this. Can anyone give some hints?

Thanks in advance.

tgingold, 2022-07-22:
There is no ROM in an fpga, so it is expected that a ROM is mapped to a BRAM.
If you RAM is not correctly described as a RAM, the synthesizer will use DFFs to implement it.