ghdl/ghdl-yosys-plugin #174: Error with altera vendor libraries

albydnc, 2022-09-26:
I'm trying to formally verify a project which has altera vendor libraries dependency. I'm able to simulate it via GHDL, but when I try to run through yosys I get this error
```
yosys> ghdl --std=08 -P=altera/ -fsynopsys slice_ops_pkg.vhd daq_frag_throttle.vhd daq_frag_throttle.psl -e daq_frag_throttle
1. Executing GHDL.
/quartus/eda/sim_lib/./220model.vhd:2463:9:error: expect wait as the first statement
        if ((lpm_representation /= "UNSIGNED") and (lpm_representation /= "SIGNED")) then
        ^
/quartus/eda/sim_lib/./220model.vhd:2258:9:error: expect wait as the first statement
        if (lpm_width <= 0) then
        ^
ERROR: vhdl import failed.
```

tgingold, 2022-09-26:
Are you sure that simlib/220model.vhd is synthesizable ?

Try --vendor-library=altera

albydnc, 2022-09-27:
I've tried it, but it did not change. I'm pretty sure it is synthesizable, since it has been already done in Quartus.

tgingold, 2022-09-27:
Thank you.  So it is just assertions.  Even if this is not compliant with 1076.6, I think I can add support for that.

closed 2022-10-02 by tgingold