ghdl/ghdl-yosys-plugin #175: Unable to build the plugin

canerbulduk, 2022-10-11:
While building the plugin via "make", I was getting the following error.

```
src/ghdl.cc:760:8: error: use of undeclared identifier 'Id_Dlatch'; did you mean 'Id_Latch'?
                case Id_Dlatch:
                     ^~~~~~~~~
                     Id_Latch
/usr/local/include/ghdl/synth_gates.h:72:4: note: 'Id_Latch' declared here
   Id_Latch = 70,
   ^
src/ghdl.cc:1093:8: error: use of undeclared identifier 'Id_Dlatch'; did you mean 'Id_Latch'?
                case Id_Dlatch:
                     ^~~~~~~~~
                     Id_Latch
/usr/local/include/ghdl/synth_gates.h:72:4: note: 'Id_Latch' declared here
   Id_Latch = 70,
   ^
2 errors generated.
make: *** [Makefile:27: ghdl.o] Error 1
```

 
```
caner@caner-VirtualBox:~$ ghdl --version
GHDL 2.0.0 (2.0.0.r0.g8185ed878) [Dunoon edition]
 Compiled with GNAT Version: 10.3.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2022 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

caner@caner-VirtualBox:~$ yosys -V
Yosys 0.22 (git sha1 f109fa3d4, gcc 9.4.0-1ubuntu1~20.04.1 -fPIC -Os)
```

As a solution, I reverted this, 90c54117cd27fcdc18ce18fb32a68062fa09e620 , commit on my end, the result was a success.

tgingold, 2022-10-11:
Get the latest checkout of ghdl or get an older version of the plugin.

janrinze, 2023-11-06:
If there was a tag that shows which version of ghdl-yosys-plugin is compatible with ghdl 2.0 that would be most useful.

The release date of ghdl 3.0.0 is march 8th 2023, your commit was already on july 14th 2022.

In my case I am working on a Debian 12 system.
The version of ghdl that comes with Debian 12 is 2.0 but no versions of ghdl-yosys-plugin support Id_Latch.
Hopefully maintainers of Debian will catch up with ghdl 3.0 soon. That will alleviate these issues.

tgingold, 2023-11-07:
This is a tricky issue as the plugin is between ghdl and yosys.  So 
ideally you'd like a tag for each release of ghdl and yosys...

Hopefully, there aren't many commits in the plugin and the interface is 
quite stable.