ghdl/ghdl-yosys-plugin #176: Error when using inout record

albydnc, 2022-10-19:
I declare an entity which has an `inout record` port.
When I run simulation in GHDL, it works as intended. When I try to run it through yosys I get this error:
```
$ yosys -m ghdl -ql ../model/design.log ../model/design.ys
shift_left.vhd:17:5:warning: no assignment for offsets 1:353 of port "snk"
    snk : inout t_comm_channel;
    ^
shift_left.vhd:18:5:warning: no assignment for offset 0 of port "src"
    src : inout t_comm_channel
    ^
ERROR: wire not found for $inout
```

tmeissner, 2022-10-19:
`Inout` ports of record types don't work with synthesis. You have to split your record in two, one for each direction.

tgingold, 2022-10-19:
I can investigate, but I need a reproducer.

albydnc, 2022-10-20:
Here you go @tgingold.
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package utils is
    type t_comm_channel is record
        valid : std_logic;
        ready : std_logic;
        data  : std_logic_vector(15 downto 0);
    end record;
end package;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use work.utils.all;

entity test is
    port (
        input : inout t_comm_channel;
        output : inout t_comm_channel
    );
end entity;

architecture rtl of test is
begin
    output <= input;
end architecture;
```
in yosys (with ghdl module loaded), run `ghdl --std=08 test.vhdl -e test`.