ghdl/ghdl-yosys-plugin #177: error: clocked logic requires clocked logic on else part

ibkvictor, 2022-11-11:
```
library IEEE;
use IEEE.std_logic_1164.all;

entity counter is
    generic( -- reusable constants in the counter logic
            Clock_freq : Integer := 50_000 ; -- high clock frequency in Hz 5000 Hz and 0.2 ms
            Time_duration : Integer := 10 ); -- stable time in ms
    port(   Clock                               : in std_logic;
            Reset                               : in std_logic;
            Switch_input                        : in std_logic; -- switch input refers to the change in signal input
            count_result                        : out std_logic:= '0'); -- checks if all the counts were fulfilled
end entity;

architecture arch of counter is
    constant num_of_counts_req : Integer := Clock_freq * Time_duration / 1000; -- (( Clock_freq * Time_duration )/ 1000); -- the 1 / 1000 converts ms to s
    signal count : Integer range 0 to num_of_counts_req - 1 := 0; -- a signal is a shared variable
    signal Switch_input_prev : std_logic;
    signal Clear : std_logic := '0';
    -- signal signal_output : std_logic := '0';

begin
    Clear <= (Switch_input_prev nand Switch_input);

    STATE_MEMORY_1 : process( Clock )
    begin
        if rising_edge(Clock) then
            Switch_input_prev <= Switch_input;
        else
            Switch_input_prev <= '0';
        end if ;
    end process ; -- STATE_MEMORY

    STATE_MEMORY_2 : process( Clock )
    begin
        if rising_edge(Clock) then
            if Reset = '1' then
                count <= 0;
                count_result <= '0';
            else 
                if Clear = '1' then
                    count <= 0;
                    -- count_result <= '0';
                    -- Reset <= '0' after 5 ms;
                else
                    if count < num_of_counts_req - 1 then
                        count <= count + 1;
                    else
                        count <= 0;
                        count_result <= Switch_input;
                    end if ;
                end if ;
            end if ;
        end if;
    end process ; -- STATE_MEMORY

    -- count_result <= signal_output;

end arch ; -- arch
```

Logic for counter in a digital debouncing circuit. However, `yosys -m ghdl -p 'ghdl --std=08 counter.vhd -e counter'` gives `ERROR: vhdl import failed.` `clocked logic requires clocked logic on else part`. I am running yosys-ghdl-plugin with a pulled  `hdlc/ghdl:yosys` container from today. Could there be a reason why I cannot synthesise the design. Thanks in advance.

tmeissner, 2022-11-11:
I think, the else branch in the clocked `if` statement is the problem:

```vhdl
    STATE_MEMORY_1 : process( Clock )
    begin
        if rising_edge(Clock) then
            Switch_input_prev <= Switch_input;
        else
            Switch_input_prev <= '0';
        end if ;
    end process ; -- STATE_MEMORY
```
What's the intention for this else branch? If you want a reset, you have to use an initial value for `Switch_input_prev` (supported in most SRAM based FPGA-architectures), or you have to use an explicit reset.

If you remove it, the design should be synthesize.

ibkvictor, 2022-11-11:
Thanks for the reply.

Without the else branch the code does not synthesize, it returns:
```
counter.vhd:27:5:error: latch infered for net "switch_input_prev" (use --latches)
    STATE_MEMORY_1 : process( Clock )
    ^
ERROR: vhdl import failed.
```

I imagine that I could include a reset, but it affects the logic as now, I would have a reset port which the device does not have. The First state memory just ensure there is previous signal to compare with on a fast clock for determining switch bouncing. Thanks once again. Is there any other thing I could do?

closed 2022-11-15 by ibkvictor