ghdl/ghdl-yosys-plugin #178: ERROR: wire not found for $posedge

ibkvictor, 2022-11-11:
Run on `Yosys 0.22+82 (git sha1 2cdbb85da, clang 11.0.1-2 -fPIC -Os)` in the docker container `hdlc/ghdl:yosys` from a recent pull. 

with the error
```
1. Executing GHDL.
Importing module counter.
ERROR: wire not found for $posedge
```

## Design
```
library IEEE;
use IEEE.std_logic_1164.all;

entity counter is
    generic( -- reusable constants in the counter logic
            Clock_freq : Integer := 50_000 ; -- high clock frequency in Hz 5000 Hz and 0.2 ms
            Time_duration : Integer := 10 ); -- stable time in ms
    port(   Clock                               : in std_logic;
            Reset                               : in std_logic;
            Switch_input                        : in std_logic; -- switch input refers to the change in signal input
            count_result                        : out std_logic:= '0'); -- checks if all the counts were fulfilled
end entity;

architecture arch of counter is
    constant num_of_counts_req : Integer := Clock_freq * Time_duration / 1000; -- (( Clock_freq * Time_duration )/ 1000); -- the 1 / 1000 converts ms to s
    signal count : Integer range 0 to num_of_counts_req - 1 := 0; -- a signal is a shared variable
    signal Switch_input_prev : std_logic;
    signal Clear : std_logic := '0';
    -- signal signal_output : std_logic := '0';

begin
    Clear <= (Switch_input_prev nand Switch_input);

    STATE_MEMORY_1 : process( Clock )
    begin
        if rising_edge(Clock) or falling_edge(Clock) then
            Switch_input_prev <= Switch_input;
        else
            Switch_input_prev <= '0';
        end if ;
    end process ; -- STATE_MEMORY

    STATE_MEMORY_2 : process( Clock )
    begin
        if rising_edge(Clock) then
            if Reset = '1' then
                count <= 0;
                count_result <= '0';
            else 
                if Clear = '1' then
                    count <= 0;
                    -- count_result <= '0';
                    -- Reset <= '0' after 5 ms;
                else
                    if count < num_of_counts_req - 1 then
                        count <= count + 1;
                    else
                        count <= 0;
                        count_result <= Switch_input;
                    end if ;
                end if ;
            end if ;
        end if;
    end process ; -- STATE_MEMORY

    -- count_result <= signal_output;

end arch ; -- arch
```

Counter logic for a debouncing circuit. The design runs in vhdl to produce the expected output `vcd` file. However, it does not synthesise in yosys. I would like to know what I need to change in the design. 
I have looked through other related issues that are closed, but those solutions seem specific to the design in question. Thanks in advance for the assistance.

PS: I am new to HDL synthesis

tmeissner, 2022-11-11:
In my experience, `wire not found for $posedge` is an error which has something to to with clocked signals. In your case, the root could be that you try to synthesize the description of an FF which is sensitive to both clock edges, which can't be handled - not only by GHDL, I know no other synthesis tools which can handle that - one reason is that FPGA HW doesn't have primitives for double data rate registers which can be mapped on that description.

So, do you really need that ddr register? Or can you rewrite your VHDL to use ordinary registers which are sensitive on one clock edge only?

ibkvictor, 2022-11-11:
Thanks a lot. I removed the falling edge in another issue. I think using another clock in that process might be a viable solution. On the other hand, removing the else path did not synthesize at all. I only added it seeing that it was good HDL practice and it was a error proceeding this one. Do you think there is something else I could do regarding the `else` path that runs into `latch infered for net "switch_input_prev" (use --latches)` error if removed while keeping the design unsynthesizable if included. Thank you once again for your prompt response.

ibkvictor, 2022-11-15:
I do not understand the entire issue, but I got it solved by removing the `falling_edge(Clock)`. Thanks to @tmeissner  for kind and prompt responses.