ghdl/ghdl-yosys-plugin #179: Error when using altera_mf lib

albydnc, 2022-11-14:
When I try to process a test entity instantiating a `scfifo` from the `altera_mf` library I get this error.

ERROR LOG:
```
ghdl --std=08 -P=altera/ -fsynopsys  test.vhdl -e test
1. Executing GHDL.
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45938:9:error: clocked logic requires clocked logic on else part
        elsif (clock'event and (clock = '0'))
        ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45221:5:error: latch infered for net "n29.max_widthu" (use --latches)
    process (clock, aclr)
    ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45221:5:error: latch infered for net "n29.numwords_minus_one" (use --latches)
    process (clock, aclr)
    ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45221:5:error: latch infered for net "n29.need_init" (use --latches)
    process (clock, aclr)
    ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45234:14:warning: variable "stratix_family" is never assigned [-Wnowrite]
    variable stratix_family         : boolean := (  FEATURE_FAMILY_STRATIX(intended_device_family) ) ;
             ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45235:14:warning: variable "showahead_area" is never assigned [-Wnowrite]
    variable showahead_area         : boolean := (lpm_showahead = "ON" and add_ram_output_register = "OFF");
             ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45236:14:warning: variable "showahead_speed" is never assigned [-Wnowrite]
    variable showahead_speed        : boolean := (lpm_showahead = "ON" and add_ram_output_register = "ON");
             ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45237:14:warning: variable "legacy_speed" is never assigned [-Wnowrite]
    variable legacy_speed           : boolean := (lpm_showahead = "OFF" and add_ram_output_register = "ON");
             ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45351:9:error: clocked logic requires clocked logic on else part
        if (clock'event and (clock = '1') and
        ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45351:9:error: clocked logic requires clocked logic on else part
        if (clock'event and (clock = '1') and
        ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45351:9:error: clocked logic requires clocked logic on else part
        if (clock'event and (clock = '1') and
        ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45351:9:error: clocked logic requires clocked logic on else part
        if (clock'event and (clock = '1') and
        ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45351:9:error: clocked logic requires clocked logic on else part
        if (clock'event and (clock = '1') and
        ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45351:9:error: clocked logic requires clocked logic on else part
        if (clock'event and (clock = '1') and
        ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45351:9:error: clocked logic requires clocked logic on else part
        if (clock'event and (clock = '1') and
        ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45351:9:error: clocked logic requires clocked logic on else part
        if (clock'event and (clock = '1') and
        ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45351:9:error: clocked logic requires clocked logic on else part
        if (clock'event and (clock = '1') and
        ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45351:9:error: synchronous code does not expect else part
        if (clock'event and (clock = '1') and
        ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45351:9:error: clocked logic requires clocked logic on else part
        if (clock'event and (clock = '1') and
        ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45351:9:error: clocked logic requires clocked logic on else part
        if (clock'event and (clock = '1') and
        ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45351:9:error: clocked logic requires clocked logic on else part
        if (clock'event and (clock = '1') and
        ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45351:9:error: clocked logic requires clocked logic on else part
        if (clock'event and (clock = '1') and
        ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45351:9:error: clocked logic requires clocked logic on else part
        if (clock'event and (clock = '1') and
        ^
/home/albertoperro/projects/scratch/fake_quartus/eda/sim_lib/./altera_mf.vhd:45351:9:error: clocked logic requires clocked logic on else part
        if (clock'event and (clock = '1') and
        ^
ERROR: vhdl import failed.
```

**minimum example**
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.math_real.all;

LIBRARY altera_mf;
USE altera_mf.altera_mf_components.all;

entity test is
  port (
    clk     : in std_logic;
    rst     : in std_logic;
    re     : in std_logic;
    we     : in std_logic;
    src_in  : in std_logic_vector(255 downto 0);
    src_out : out std_logic_vector(255 downto 0)
  );
end entity;

architecture rtl of test is

begin

  scfifo_rx : entity altera_mf.scfifo
  generic map (
    lpm_width => 256,
    lpm_widthu => 2,
    lpm_numwords => 2,
    lpm_showahead => "OFF",
    lpm_hint => "USE_EAB=ON",
    ram_block_type => "AUTO",
    intended_device_family => "Arria 10",
    almost_full_value => 0,
    almost_empty_value => 0,
    overflow_checking => "ON",
    underflow_checking => "ON",
    allow_rwcycle_when_full => "OFF",
    add_ram_output_register => "OFF",
    use_eab => "ON",
    lpm_type => "scfifo",
    enable_ecc => "false",
    maximum_depth => 0
  )
  port map (
    aclr => rst,
    clock => clk,
    wrreq => we,
    data => src_in,
    full => open,
    rdreq => re,
    q => src_out,
    empty => open,
    usedw => open
  );

end architecture;
```

tgingold, 2022-11-14:
This scfifo is a weird as it uses double clocked process.  I need to understand it first.

tgingold, 2022-11-15:
I am not sure what could be done, except improving the error message.
scfifo doesn't look to be synthesizable.  Maybe Quartus handles it as a special case.
The way it set the initial value is weird and certainly not conformant with 1076.6
Some signals are assigned on both clock edge.

I fear the way to move forward is to slowly rewrite altera_mf in a fully synthesizable style.

Would be interesting to know how it is handled by quartus or symplify.

tmeissner, 2023-03-08:
I assume that the errors occurs because of you are trying to synthesize a multi-clock FF, which is not supported by yosys. See #127 for a similar issue.