ghdl/ghdl-yosys-plugin #181: Linking against libgnat-9.so.1

jjjt-git, 2022-12-04:
When I build the plugin on my system (gentoo) with standart options (./configure && make && make install), it will build fine, but when I try to use the plugin (plugin -i ghdl) it will fail, because the system is unable to find libgnat-9.
That behaviour is expected, because on my system there only is libgnat-11.
So the question is, if it would be possible to link against the system library.

tgingold, 2022-12-07:
I am not sure to understand.  If you use the system compiler it should use the system library.

Which compiler are you using ?
What is the output of 'gnatls -v' ?

jjjt-git, 2022-12-07:
gnatls -v:
GNATLS 11.3.0
Copyright (C) 1997-2021, Free Software Foundation, Inc.

Source Search Path:
   <Current_Directory>
   /usr/lib/gcc/x86_64-pc-linux-gnu/11.3.0/adainclude

Object Search Path:
   <Current_Directory>
   /usr/lib/gcc/x86_64-pc-linux-gnu/11.3.0/adalib

Project Search Path:
   <Current_Directory>
   /usr/x86_64-pc-linux-gnu/lib/gnat
   /usr/x86_64-pc-linux-gnu/share/gpr
   /usr/share/gpr
   /usr/lib/gnat

tgingold, 2022-12-08:
Ok; are you also building ghdl and libghdl ?

jjjt-git, 2022-12-08:
Yes, I did build ghdl and libghdl on the same System. libghdl is linked against libgnat-11.so

tgingold, 2022-12-08:
So which library is linked against libgnat-9.so ?

jjjt-git, 2022-12-08:
The ghdl.so built by make.
In directory ghdl-yosys-plugin:
$ ldd ghdl.so
        linux-vdso.so.1 (0x00007ffe755db000)
        libghdl-3_0_0_dev.so => /<HOME>/Projects/f4pga/f4pga/xc7/conda/envs/xc7/lib/libghdl-3_0_0_dev.so (0x00007fa8921cc000)
        libstdc++.so.6 => /<HOME>/Projects/f4pga/f4pga/xc7/conda/envs/xc7/lib/libstdc++.so.6 (0x00007fa891fb8000)
        libgcc_s.so.1 => /<HOME>/Projects/f4pga/f4pga/xc7/conda/envs/xc7/lib/libgcc_s.so.1 (0x00007fa891f9e000)
        libc.so.6 => /lib64/libc.so.6 (0x00007fa891d7d000)
        libm.so.6 => /lib64/libm.so.6 (0x00007fa891ca0000)
        libgnat-9.so.1 => /<HOME>/Projects/f4pga/f4pga/xc7/conda/envs/xc7/lib/libgnat-9.so.1 (0x00007fa89192a000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fa892609000)
        libdl.so.2 => /lib64/libdl.so.2 (0x00007fa891925000)

jjjt-git, 2022-12-08:
Most curiously, on another computer, where I installed the plugin in the same way, it was linked against libgnat-11.so (system).

tgingold, 2022-12-09:
> libgnat-9.so.1 => //Projects/f4pga/f4pga/xc7/conda/envs/xc7/lib/libgnat-9.so.1 (0x00007fa89192a000)

Well, you don't compile with the system compiler!

jjjt-git, 2022-12-09:
In the background it is using the system-compiler (simple symlink).
But I will analyse what went wrong and maybe it will resolve itself.

jjjt-git, 2022-12-11:
Ok, it solved itself.
I had compiled by accident against a wrong include- and libdir.

closed 2022-12-11 by jjjt-git