ghdl/ghdl-yosys-plugin #182: ERROR: No such command: ghdl (type 'help' for a command overview)

wklam296, 2022-12-17:
I can't use the plugin, it appears error ERROR: No such command: ghdl (type 'help' for a command overview) after following the installing instruction.
How can I solve the problem?

wklam296, 2022-12-17:
yosys  \
  -p \
  "ghdl --std=08 --workdir=build -Pbuild --no-formal neorv32_iCEBreaker_BoardTop_UP5KDemo; \
   synth_ice40 \
  -top neorv32_iCEBreaker_BoardTop_UP5KDemo -dsp \
  -json neorv32_iCEBreaker_UP5KDemo.json" 2>&1 | tee yosys-report.txt
ERROR: No such command: ghdl (type 'help' for a command overview)

tgingold, 2022-12-17:
It's a plugin you need to load. Try: `yosys -m ghdl ...`

wklam296, 2022-12-19:
yosys  \
  -m \
  "ghdl --std=08 --workdir=build -Pbuild --no-formal neorv32_iCEBreaker_BoardTop_UP5KDemo; \
   synth_ice40 \
  -top neorv32_iCEBreaker_BoardTop_UP5KDemo -dsp \
  -json neorv32_iCEBreaker_UP5KDemo.json" 2>&1 | tee yosys-report.txt

ERROR: Can't load module `./ghdl --std=08 --workdir=build -Pbuild --no-formal neorv32_iCEBreaker_BoardTop_UP5KDemo;    synth_ice40   -top neorv32_iCEBreaker_BoardTop_UP5KDemo -dsp   -json neorv32_iCEBreaker_UP5KDemo.json':

umarcor, 2022-12-19:
@wklam296 see https://github.com/stnolting/neorv32-setups/blob/main/osflow/common.mk#L20 and https://github.com/stnolting/neorv32-setups/blob/main/osflow/synthesis.mk#L18.

closed 2022-12-20 by wklam296