ghdl/ghdl-yosys-plugin #184: ECP5 Example fails: 'VLO' is unsupported

playduck, 2023-04-17:
I'm trying to build the [ECP5_Versa example](https://github.com/ghdl/ghdl-yosys-plugin/tree/master/examples/ecp5_versa) in this repository.

Yosys and this plugin run fine. Nextpnr however throws the following error:
```
ERROR: cell type 'VLO' is unsupported (instantiated as 'clk_pll1.scuba_vlo_inst')
```

I've only modified the makefile to not use docker and use the local versions of tools instead.
These tools are:
- `GHDL 4.0.0-dev (3.0.0.r66.g776731910) [Dunoon edition]
 Compiled with GNAT Version: 10.4.0
 llvm 14.0.0 code generator`
- `Yosys 0.28+1 (git sha1 a9c792dce, clang 14.0.0-1ubuntu1 -fPIC -Os)`
- `"nextpnr-ecp5" -- Next Generation Place and Route (Version nextpnr-0.5-107-g71192dc7)`
- ghdl-yosys-plugin git hash `7a71af5845b0789d2a8169336d0e07a4c27beb1d` (current master as of 2023-04-16)

on Ubuntu 22.04.2 LTS in a virtual machine (UTM on MacOs / aarch64).

I've noticed that `VLO` and `VHI` are implemented as primitives in [`/library/wrapper/primitives.v`](https://github.com/ghdl/ghdl-yosys-plugin/blob/master/library/wrapper/primitives.v). These are also present on my system.
Furthermore yosys says:
```
...
2. Executing Verilog-2005 frontend: ../../library/wrapper/primitives.v
Parsing Verilog input from `../../library/wrapper/primitives.v' to AST representation.
Generating RTLIL representation for module `\vhi'.
Generating RTLIL representation for module `\vlo'.
Successfully finished Verilog frontend.
...
```
At the end it successfully prints the usage statistics including
```
     VLO                             1
```
The VLO module is also present in the .json

Pastebin of `make clean all` logs: https://pastebin.com/ePc43MTa

---

I've built all tools from source and I don't specifically remember building or installing a Verilog compiler myself.
However if I was missing that I'd assume yosys would complain along the way.

Is this an issue on my end, with this plugin/example or with nextpnr?

tgingold, 2023-04-18:
I haven't tested with nextpnr, but can to try to convert to uppercase vli and vlo in primitives.v ?

closed 2023-04-18 by playduck

tgingold, 2023-04-18:
Yes please!