ghdl/ghdl-yosys-plugin #187: An error occurred while building the plugin

48004800, 2023-04-30:
I can successfully build both ghdl and yosys. However, a large number of "undefined reference to.....'"errors may occur when building plugins.
Because my operating system is Windows. I built the plugin using the second method. （Build as part of yosys）

When I try to build yosys, the following error occurs.
[image]

tgingold, 2023-04-30:
You need to add '-lgnat' to the linker flags.  Maybe do:
make LDFLAGS=-lgnat

48004800, 2023-05-01:
> You need to add '-lgnat' to the linker flags. Maybe do: make LDFLAGS=-lgnat

cannot find - lgnat

I built yosys separately through Cygwin. But I need to remove the - fPIC in the Makefile, and I'm not sure if 'undefined reference to……' is related to this.

48004800, 2023-05-01:
I don't have much experience in this area, I have already spent two days building ghdl yosys plugin. I have decided to give up.

48004800, 2023-05-26:
> You need to add '-lgnat' to the linker flags. Maybe do: make LDFLAGS=-lgnat

I want to ask a simple question first. If I want to build this plugin on Windows, what tools do I need in the step of building yosys.

tgingold, 2023-05-26:
In addition to a C++ compiler, you need to install the gnat compiler.
On mingw64, just do: pacman -S mingw-w64-x86_64-gcc-ada

48004800, 2023-05-27:
> In addition to a C++ compiler, you need to install the gnat compiler. On mingw64, just do: pacman -S mingw-w64-x86_64-gcc-ada

What is the reason for this error？

 ‘strdup’ was not declared in this scope; did you mean ‘strcmp’?

tgingold, 2023-05-27:
Probably a missing #include; what is the context of the error ?

48004800, 2023-05-27:
> Probably a missing #include; what is the context of the error ?

I have solved it now by modifying the makefile and changing c++11 to gnu++11.

48004800, 2023-05-27:
> Probably a missing #include; what is the context of the error ?

I have now built yosys separately using gcc. I can build GHDL and yosys separately. But I can't build plugins

[message.txt](https://github.com/ghdl/ghdl-yosys-plugin/files/11581622/message.txt)

umarcor, 2023-05-27:
It's not possible to build the plugin as a shared library (module) on Windows, because yosys does not support loading them on that platform. It needs to be built-in.

48004800, 2023-05-27:
> It's not possible to build the plugin as a shared library (module) on Windows, because yosys does not support loading them on that platform. It needs to be built-in.

[image]

I am indeed trying to follow this step. Which step did I take wrong?

tgingold, 2023-05-27:
None, but the instructions are old and mainly unused.

You need to link with -lgnat

48004800, 2023-05-27:
> None, but the instructions are old and mainly unused. You need to link with -lgnat

I don't know exactly what's missing

$ make LDFLAGS=-lgnat
[Makefile.conf] CONFIG := gcc
[Makefile.conf] ENABLE_GHDL := 1
[Makefile.conf] GHDL_PREFIX := "C:\ghdl-3.0.0\build\PREFIX"
[100%] Building yosys
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: cannot find -lgnat: No such file or directory
collect2: error: ld returned 1 exit status
make: *** [Makefile:743: yosys] Error 1

tgingold, 2023-05-27:
Ok, in "C:\ghdl-3.0.0\build\PREFIX\lib" there should be a libghdl.link file.
What is the content of it ?

48004800, 2023-05-27:
> Ok, in "C:\ghdl-3.0.0\build\PREFIX\lib" there should be a libghdl.link file. What is the content of it ?

C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a

tgingold, 2023-05-27:
Ok, that's where is libgnat.

However, I think there is a missing ')' in the yosys Makefile, at the end of this line:

LDLIBS += $(GHDL_LIB_DIR)/libghdl.a $(file <$(GHDL_LIB_DIR)/libghdl.link)

I suppose it will link correctly if you just add a ')'.

48004800, 2023-05-27:
[image]

tgingold, 2023-05-27:
So yes, there is a missing ')'.

48004800, 2023-05-28:
> So yes, there is a missing ')'.

[image]
This seems incorrect

tgingold, 2023-05-28:
Oops sorry.  What is your version of make ?

48004800, 2023-05-28:
> Oops sorry. What is your version of make ?

4.4.1-1

tgingold, 2023-05-28:
This is the last one.

What is the link command displayed by make ?

48004800, 2023-05-28:
> This is the last one. What is the link command displayed by make ?

Sorry, I don't understand what a link command is.

48004800, 2023-05-28:
> This is the last one. What is the link command displayed by make ?

Complete output.

[message.txt](https://github.com/ghdl/ghdl-yosys-plugin/files/11583882/message.txt)

tgingold, 2023-05-28:
You need to use 'make PRETTY=0' to get a verbose but complete output.

48004800, 2023-05-28:
[message.txt](https://github.com/ghdl/ghdl-yosys-plugin/files/11583897/message.txt)

tgingold, 2023-05-28:
In the Makefile, can you replace

$(file <$(GHDL_LIB_DIR)/libghdl.link)

with

$(shell cat $(GHDL_LIB_DIR)/libghdl.link)

48004800, 2023-05-28:
> In the Makefile, can you replace $(file <$(GHDL_LIB_DIR)/libghdl.link) with $(shell cat $(GHDL_LIB_DIR)/libghdl.link)

I'm trying now.

$ make
[Makefile.conf] CONFIG := gcc
[Makefile.conf] ENABLE_GHDL := 1
[Makefile.conf] GHDL_PREFIX := "C:\ghdl-3.0.0\build\PREFIX"
[  0%] Building kernel/version_9c5a60eb201.cc
[  0%] Building kernel/version_9c5a60eb201.o
[100%] Building yosys
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:\ghdl-3.0.0\build\PREFIX/lib/libghdl.a(grt-cstdio.o): in function `fprintf':
C:\ghdl-3.0.0\build/C:/ghdl-3.0.0/src/grt/grt-cstdio.c:364: undefined reference to `__mingw_vfprintf'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:\ghdl-3.0.0\build\PREFIX/lib/libghdl.a(grt-cstdio.o): in function `snprintf':
C:\ghdl-3.0.0\build/C:/ghdl-3.0.0/src/grt/grt-cstdio.c:462: undefined reference to `__mingw_vsnprintf'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:\ghdl-3.0.0\build\PREFIX/lib/libghdl.a(grt-cstdio.o): in function `__ghdl_get_stdout':
C:\ghdl-3.0.0\build/C:/ghdl-3.0.0/src/grt/grt-cstdio.c:28: undefined reference to `__imp___acrt_iob_func'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:\ghdl-3.0.0\build\PREFIX/lib/libghdl.a(grt-cstdio.o): in function `__ghdl_get_stdin':
C:\ghdl-3.0.0\build/C:/ghdl-3.0.0/src/grt/grt-cstdio.c:34: undefined reference to `__imp___acrt_iob_func'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:\ghdl-3.0.0\build\PREFIX/lib/libghdl.a(grt-cstdio.o): in function `__ghdl_get_stderr':
C:\ghdl-3.0.0\build/C:/ghdl-3.0.0/src/grt/grt-cstdio.c:40: undefined reference to `__imp___acrt_iob_func'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x4e): undefined reference to `__imp__wsplitpath'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x171): undefined reference to `__imp_wcscat_s'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x1b1): undefined reference to `__imp_wcscat_s'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x1e3): undefined reference to `__imp_wcscat_s'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x575): undefined reference to `__imp__time64'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x59e): undefined reference to `__imp__time64'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x5ae): undefined reference to `__imp__localtime64'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x63e): undefined reference to `__imp__gmtime64'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x723): undefined reference to `__imp__mktime64'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x7fd): undefined reference to `__imp_wcscpy_s'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x808): undefined reference to `__imp_wcscat_s'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x844): undefined reference to `__imp__wopen'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x951): undefined reference to `__imp__wgetcwd'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0xb0b): undefined reference to `__imp__wunlink'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0xbb2): undefined reference to `__imp__wrename'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0xc2b): undefined reference to `__imp__wchdir'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0xc9b): undefined reference to `__imp__wrmdir'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0xd65): undefined reference to `__imp__wfopen'
[... cut ...]
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(raise-gcc.o):(.text+0x262): undefined reference to `__imp___acrt_iob_func'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(raise-gcc.o):(.text+0x2a8): undefined reference to `__mingw_vfprintf'
collect2: error: ld returned 1 exit status
make: *** [Makefile:743: yosys] Error 1

48004800, 2023-05-29:
```
$ make
[Makefile.conf] ENABLE_GHDL := 1
[Makefile.conf] GHDL_PREFIX := "C:\ghdl-3.0.0\build\PREFIX"
[Makefile.conf] CONFIG := gcc
[100%] Building yosys
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:\ghdl-3.0.0\build\PREFIX/lib/libghdl.a(grt-cstdio.o): in function `fprintf':
C:\ghdl-3.0.0\build/C:/ghdl-3.0.0/src/grt/grt-cstdio.c:364: undefined reference to `__mingw_vfprintf'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:\ghdl-3.0.0\build\PREFIX/lib/libghdl.a(grt-cstdio.o): in function `snprintf':
C:\ghdl-3.0.0\build/C:/ghdl-3.0.0/src/grt/grt-cstdio.c:462: undefined reference to `__mingw_vsnprintf'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:\ghdl-3.0.0\build\PREFIX/lib/libghdl.a(grt-cstdio.o): in function `__ghdl_get_stdout':
C:\ghdl-3.0.0\build/C:/ghdl-3.0.0/src/grt/grt-cstdio.c:28: undefined reference to `__imp___acrt_iob_func'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:\ghdl-3.0.0\build\PREFIX/lib/libghdl.a(grt-cstdio.o): in function `__ghdl_get_stdin':
C:\ghdl-3.0.0\build/C:/ghdl-3.0.0/src/grt/grt-cstdio.c:34: undefined reference to `__imp___acrt_iob_func'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:\ghdl-3.0.0\build\PREFIX/lib/libghdl.a(grt-cstdio.o): in function `__ghdl_get_stderr':
C:\ghdl-3.0.0\build/C:/ghdl-3.0.0/src/grt/grt-cstdio.c:40: undefined reference to `__imp___acrt_iob_func'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x4e): undefined reference to `__imp__wsplitpath'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x171): undefined reference to `__imp_wcscat_s'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x1b1): undefined reference to `__imp_wcscat_s'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x1e3): undefined reference to `__imp_wcscat_s'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x575): undefined reference to `__imp__time64'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x59e): undefined reference to `__imp__time64'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x5ae): undefined reference to `__imp__localtime64'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x63e): undefined reference to `__imp__gmtime64'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x723): undefined reference to `__imp__mktime64'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x7fd): undefined reference to `__imp_wcscpy_s'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x808): undefined reference to `__imp_wcscat_s'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x844): undefined reference to `__imp__wopen'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0x951): undefined reference to `__imp__wgetcwd'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0xb0b): undefined reference to `__imp__wunlink'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0xbb2): undefined reference to `__imp__wrename'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0xc2b): undefined reference to `__imp__wchdir'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0xc9b): undefined reference to `__imp__wrmdir'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0xd65): undefined reference to `__imp__wfopen'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(adaint.o):(.text+0xe4f): undefined reference to `__imp__wfreopen'
[... cut ...]
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(raise-gcc.o):(.text+0x262): undefined reference to `__imp___acrt_iob_func'
/usr/lib/gcc/x86_64-pc-msys/11.3.0/../../../../x86_64-pc-msys/bin/ld: C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/13.1.0/adalib/libgnat.a(raise-gcc.o):(.text+0x2a8): undefined reference to `__mingw_vfprintf'
collect2: error: ld returned 1 exit status
make: *** [Makefile:743: yosys] Error 1
```

tgingold, 2023-05-29:
That's much better, but it looks like you are compiling yosys for msys but ghdl for mingw.
They are not compatible.

48004800, 2023-05-29:
> That's much better, but it looks like you are compiling yosys for msys but ghdl for mingw. They are not compatible.

So do I need to use msys2 mingw64?

48004800, 2023-05-29:
> That's much better, but it looks like you are compiling yosys for msys but ghdl for mingw. They are not compatible.

I am currently using msys2 UCRT64.

QQ109@Mine UCRT64 /c/yosys-yosys-0.29
$ make
[Makefile.conf] ENABLE_GHDL := 1
[Makefile.conf] GHDL_PREFIX := "C:\ghdl-3.0.0\build\PREFIX"
[Makefile.conf] CONFIG := gcc
[100%] Building yosys
gcc.exe: error: unrecognized command-line option '-rdynamic'
make: *** [Makefile:743: yosys] Error 1

closed 2023-05-29 by 48004800