ghdl/ghdl-yosys-plugin #192: wire not found for $posedge

GyrosGeier, 2023-11-12:
Completely different code for the other issue with the same title, so a separate issue. At least this one is nice and minimal.

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity test is
        port(
                clk, set, clear, reset : in std_logic;
                o : out std_logic
        );
end entity;

architecture rtl of test is
begin
        o <=
                '0'     when ?? reset else
                '1'     when (?? set) and rising_edge(clk) else
                '0'     when (?? clear) and rising_edge(clk);
end architecture;
```

This falls over with "wire not found for $posedge", while trying to connect the `rising_edge` as an input to an `and`, because the `get_src` function does not handle `Id_PosEdge`.

Writing this as

```vhdl
        o <=
                '0'                      when ?? reset else
                set or (o and not clear) when rising_edge(clk);
````
works, as do normal enables like

```vhdl
        addr_r <= addr when (?? valid) and rising_edge(clk);
```
so at least *some* variants of this are resolved by GHDL before they reach the GHDL YoSYS plugin.

GyrosGeier, 2023-11-12:
If the LHS of an assignment with an enable is a subset, this also falls over:

```vhdl
    signal s : std_logic_vector(31 downto 0);
...
    s(15 downto 0) <= i when (?? valid) and rising_edge(clk);
```

tgingold, 2023-11-12:
I am not sure your construct is allowed by 1076.6
I will have a look.

simonjrichter, 2023-11-13:
Hmm, even if it is invalid it would be nice to have better diagnostics here -- after all it simulates fine in GHDL, and "connecting a `$posedge` to something that doesn't explicitly expect it and knows not to call `get_src` on it" seems to be an entire class of errors.

GyrosGeier, 2023-11-13:
Hmm, even if it is invalid it would be nice to have better diagnostics here -- after all it simulates fine in GHDL, and "connecting a `$posedge` to something that doesn't explicitly expect it and knows not to call `get_src` on it" seems to be an entire class of errors.