ghdl/ghdl-yosys-plugin #195: libgnat-13.so module missing

95lux, 2024-05-21:
I am trying to load the module with `yosys -m ghdl -p 'ghdl'` but get

`ERROR: Can't load module `./ghdl': libgnat-13.so: cannot open shared object file: No such file or directory`

I am on arch, and when trying to install libgnat with ada-gcc package, it only installs the libgnat-14.so library object.
I also tried building yosys, ghdl-yosys-plugin and ghdl from Github source, to rule out the possibility of outdated, or misconfigured packages form the AUR.

Is the dependency of libgnat13 correct?

Xiretza, 2024-05-21:
You built ghdl while GCC 13 was current. Now, GCC 14 is current. Rebuild ghdl.

95lux, 2024-05-21:
Thanks alot! I recompiled ghdl, and yosys plugin. Now i got a working installation again!
Sorry for opening an issue for it.

closed 2024-05-21 by 95lux