ghdl/ghdl-yosys-plugin #197: How to run unisim using cxxrtl

aniketabhiraj2004, 2024-06-24:


aniketabhiraj2004, 2024-06-24:
yosys -m ghdl -p "ghdl --std=93c --ieee=synopsys -fexplicit -Whide -Wspecs dut2; write_cxxrtl blink.cpp"

 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2024  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.42+28 (git sha1 ede3750a6, g++ 10.2.0 -fPIC -Os)

-- Running command `ghdl --std=93c --ieee=synopsys -fexplicit -Whide -Wspecs dut2; write_cxxrtl blink.cpp' --

1. Executing GHDL.
/tools/Xilinx/Vivado/2019.2/data/vhdl/src/unisims/primitive/FIFO18E2.vhd:420:27:error: no declaration for "gsr"
    glblGSR     <= TO_X01(GSR);
                          ^
ERROR: vhdl import failed.

tgingold, 2024-06-25:
Apparently, there are some global signals which aren't yet supported.  You could probably try to modify unisim to remove them.

If you have a reproducer (not using unisim, which is not open source), I could try to add support for it.

aniketabhiraj2004, 2024-06-25:
Can you suggest some open source reproducer.

aniketabhiraj2004, 2024-06-26:
> Can you suggest some open source reproducer.

Xiretza, 2024-06-26:
> Can you suggest some open source reproducer.

That's what's being asked of *you*. For the issue to be fixed, there needs to be a design that reproduces the issue without using proprietary libraries - you're in the best position to create such a design, since you just need to modify your existing reproducer to not use unisim.

befedo, 2024-06-26:
> yosys -m ghdl -p "ghdl --std=93c --ieee=synopsys -fexplicit -Whide -Wspecs dut2; write_cxxrtl blink.cpp"
> 
> /---------------------------------------------------------------------------- | yosys -- Yosys Open SYnthesis Suite | | Copyright (C) 2012 - 2024 Claire Xenia Wolf [claire@yosyshq.com](mailto:claire@yosyshq.com) | | Distributed under an ISC-like license, type "license" to see terms | ----------------------------------------------------------------------------/ Yosys 0.42+28 (git sha1 ede3750a6, g++ 10.2.0 -fPIC -Os)
> 
> -- Running command `ghdl --std=93c --ieee=synopsys -fexplicit -Whide -Wspecs dut2; write_cxxrtl blink.cpp' --
> 
> 1. Executing GHDL.
>    /tools/Xilinx/Vivado/2019.2/data/vhdl/src/unisims/primitive/FIFO18E2.vhd:420:27:error: no declaration for "gsr"
>    glblGSR     <= TO_X01(GSR);
>    ^
>    ERROR: vhdl import failed.

If I remember correctly, the GSR/GTS signals are only necessary for simulation and where generated like shown [here](https://docs.amd.com/r/en-US/ug900-vivado-logic-simulation/Global-Set-and-Reset-and-Global-3-State-Signals-in-VHDL).

Did you include `GLBL_VHD.vhd`?

aniketabhiraj2004, 2024-07-15:
Is signal in package is allowed in vhdl.

tgingold, 2024-07-16:
Yes, signal declarations are allowed in package according to the VHDL LRM.
But I fear they aren't in the synthesizable subset.