ghdl/ghdl-yosys-plugin #203: Get Verilog netlist

GTP95, 2024-12-02:
Greetings,

Sorry if my question is naive, I don't have experience with circuit synthesis tools. By default, Yosys doesn't support VHDL, but it is possible to add support via this plugin. However, when synthetizing a VHDL design via this plugin, I get as output a JSON and a `.cf` file. My question is, would it be possible to get a Verilog netlist instead? I need to use it as input for another tool.
Thank you for your help,

GTP

tmeissner, 2024-12-02:
You can find the appropriate commands in the GHDL documentation: https://ghdl.github.io/ghdl/using/Synthesis.html#convert-v-hdl-to-other-formats

GTP95, 2024-12-04:
Thank you for your prompt answer. Unfortunately, I'm still facing issues as the [project](https://github.com/hgrosz/aes-dom) I'm trying to convert is composed of multiple files. I'm able to synthesize it if I first run `ghdl -a` on the project files in the correct order, like this:
```
ghdl -a masked_aes_pkg.vhdl
ghdl -a aes_ctrl_lsfr_23_cycles.vhdl
ghdl -a rcon.vhdl
ghdl -a lin_map.vhdl
ghdl -a square_scaler.vhdl
ghdl -a gf2_mul.vhdl
ghdl -a shared_mul_gf4.vhdl
ghdl -a shared_mul_gf2.vhdl
ghdl -a real_dom_shared_mul_gf2.vhdl
ghdl -a inverter.vhdl
ghdl -a real_dom_shared_mul_gf4.vhdl
ghdl -a aes_sbox.vhdl
ghdl -a aes_key_regs.vhdl
ghdl -a mix_columns.vhdl
ghdl -a aes_state_regs.vhdl
ghdl -a aes_ctrl_lsfr.vhdl
ghdl -a aes_top.vhdl
```
and then run `yosys -m ghdl -p 'ghdl aes_top; synth_ice40 -json aes_top.json'`.
But if I try the command I found in the documentation, I get the following outputs:

```
$ yosys -m ghdl -p 'ghdl aes_top.vhdl -e aes_top; write_verilog Verilog/aes-dom.v'

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.33 (git sha1 2584903a060)

-- Running command `ghdl aes_top.vhdl -e aes_top; write_verilog Verilog/aes-dom.v' --

1. Executing GHDL.
aes_top.vhdl:48:25:error: unit "masked_aes_pkg" not found in library "work"
               use work.masked_aes_pkg.all;
                        ^
aes_top.vhdl:62:19:error: no declaration for "t_shared_gf8"
    PTxDI    : in t_shared_gf8(N downto 0);
                  ^
aes_top.vhdl:64:19:error: no declaration for "t_shared_gf8"
    KxDI     : in t_shared_gf8(N downto 0);
                  ^
aes_top.vhdl:66:19:error: no declaration for "t_shared_gf4"
    Zmul1xDI : in t_shared_gf4((N*(N+1)/2)-1 downto 0);  -- for y1 * y0
                  ^
aes_top.vhdl:67:19:error: no declaration for "t_shared_gf4"
    Zmul2xDI : in t_shared_gf4((N*(N+1)/2)-1 downto 0);  -- for O * y1
                  ^
aes_top.vhdl:68:19:error: no declaration for "t_shared_gf4"
    Zmul3xDI : in t_shared_gf4((N*(N+1)/2)-1 downto 0);  -- for O * y0
                  ^
aes_top.vhdl:69:19:error: no declaration for "t_shared_gf2"
[... 36 lines cut ...]
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.33 (git sha1 2584903a060)

-- Running command `ghdl masked_aes_pkg.vhdl -e aes_top; write_verilog masked_aes_pkg.v' --

1. Executing GHDL.
error: cannot find entity or configuration aes_top
ERROR: vhdl import failed.
```
How should I proceed in the case of a project split into multiple files?
Thank you for your help,

GTP

tmeissner, 2024-12-04:
These commands doesn't match:

```
yosys -m ghdl -p 'ghdl aes_top; synth_ice40 -json aes_top.json'.
```
and in the yosys log:
```
$ yosys -m ghdl -p 'ghdl aes_top.vhdl -e aes_top; write_verilog Verilog/aes-dom.v'
```

No need to compile the VHDL source files again within yosys.

I think if you really use `yosys -m ghdl -p 'ghdl aes_top; synth_ice40 -json aes_top.json'` it will work.  Or something like `yosys -m ghdl -p 'ghdl aes_top; write_verilog Verilog/aes-dom.v` to get the Verilog netlist.

GTP95, 2024-12-04:
Yes, i know that the two commands are meant to do different things. I just wanted to show that I'm running the commands on a working project. After cleaning up the synthesis artifacts, I tried the command that you suggested. But it didn't work:

```
$ yosys -m ghdl -p 'ghdl aes_top; write_verilog Verilog/aes-dom.v'

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.33 (git sha1 2584903a060)

-- Running command `ghdl aes_top; write_verilog Verilog/aes-dom.v' --

1. Executing GHDL.
error: cannot find entity or configuration aes_top
ERROR: vhdl import failed.
```
Maybe I'm specifying the wrong name for the entity?

tgingold, 2024-12-04:
@GTP95, to clarify.  Either you analyze the files (outside yosys) and then synthesize using the top level:
```bash
ghdl -a file1.vhdl
ghdl -a file2.vhdl
...
ghdl -a top.vhdl
yosys -m ghdl -p 'ghdl -e top; ...'
```

Or  you give directly all the files to ghdl:
```bash
yosys -m ghdl -p 'ghdl file1.vhdl file2.vhdl ... top.vhdl -e top; ...'
```

GTP95, 2024-12-05:
Thank you for the clarification. I tried analyzing all the files and then synthesizing the top module, and got `error: unknown command option '-e'` from GHDL. But the second method (passing all files to GHDL) worked. Thank you!

closed 2024-12-05 by GTP95