ghdl/ghdl-yosys-plugin #204: "ERROR: Assert `chunk.width != 0' failed in kernel/rtlil.cc:5012." with an "integer range 0 to 0" inside a record

eduardosm, 2024-12-05:
i'm trying to synthesize the following reduced example: 

`defs.vhd`:

```vhdl
package defs is
    type rec is record
        x: integer range 0 to 1;
        y: integer range 0 to 0;
        z: boolean;
    end record;
end package;
```

`dummy.vhd`:

```vhdl
entity dummy is
    port (
        o: out work.defs.rec
    );
end entity;

architecture arch of dummy is
begin
end architecture;
```

`top_level.vhd`:
```vhdl
entity top_level is
end entity;

architecture arch of top_level is
    signal s: work.defs.rec;
begin
    inst: entity work.dummy
    port map (
        o => s
    );
end architecture;
```

with

```sh
mkdir out
ghdl analyze --std=93 --work=top --workdir="out" -P"out" defs.vhd dummy.vhd top_level.vhd
yosys -m "$GHDL_YOSYS" -p "ghdl --std=93 --work=top --workdir=out -Pout top_level"
```

(assuming `GHDL_YOSYS` is set to the path of the plugin library)

It fails with the following output:

```
-- Running command `ghdl --std=93 --work=top --workdir=out -Pout top_level' --

1. Executing GHDL.
dummy.vhd:3:9:warning: no assignment for port "o" [-Wnowrite]
        o: out work.defs.rec
        ^
Importing module top_level.
Importing module dummy.
ERROR: Assert `chunk.width != 0' failed in kernel/rtlil.cc:5012.
```

I have GHDL 3.0.0-1.20230308git7de967c, Yosys 0.47-1.20241110gitf20f913 (from Fedora 41 repository) and ghdl-yosys-plugin built from commit 511412f984d64ed7c46c4bdbd839f4b3c48f6fa5.

tgingold, 2024-12-06:
Ok, clearly the issue is the net `rec.y` which requires 0 bit.

eduardosm, 2024-12-06:
Simpler example (single file):
```vhd
package defs is
    type rec is record
        a: integer range 0 to 1;
        b: integer range 0 to 0;
        c: integer range 0 to 1;
    end record;
end package;

entity top_level is
    port (
        o: out work.defs.rec
    );
end entity;

architecture arch of top_level is
begin
    o <= (
        a => 0,
        b => 0,
        c => 0
    );
end architecture;
```

The issue might be in GHDL itself instead of this plugin. I tried `ghdl synth --out=verilog --std=93 top_level.vhd -e top_level` to produce Verilog without involving this plugin or Yosys and it produced invalid syntax. Reported at https://github.com/ghdl/ghdl/issues/2826.

eduardosm, 2024-12-15:
Fixed in GHDL (https://github.com/ghdl/ghdl/issues/2826, https://github.com/ghdl/ghdl/issues/2827, https://github.com/ghdl/ghdl/issues/2834).

closed 2024-12-15 by eduardosm