ghdl/ghdl-yosys-plugin #210: vhdl asserts not stripped

nathanaelhuffman, 2025-02-11:
To be honest, I'm not 100% sure where this should go and if this is even necessarily a GHDL plugin bug, but figuring out what was going on was a bit tricky.  I have this code in a function in a package, to mostly help trip an assert in simulation. I realize runtime asserts don't make any real sense in an implemented hw design, and in my case, there's logic that guarantees only the first two cases can be hit in my hw design.  This is doubly-tricky b/c this is a run-time assert, not something that can be determined by elaboration.  My maybe naive expectation was that this assert to be silently dropped somewhere along the way to a bitstream, but that is not what is happening.
 
```
case opcode is
            when spi_opcode_bit_set =>
                return rdata or wdata;
            when spi_opcode_bit_clr =>
                return rdata and (not wdata);
            when others =>
                assert false report "not a bit operation opcode" severity failure; -- <-- Commenting out this line works fine.
                return rdata;
        end case;
```
This assert is passed through the GHDL plugin to yosys, which passes it on into the .json, but then the place&route backend trips out with 
something like:
`ERROR: cell type '$assert' is unsupported (instantiated as 'spi_axi_controller_inst.spi_axi_sms_inst.679')`

because it was passed in a cell that looks like this:
```
"spi_axi_controller_inst.spi_axi_sms_inst.679": {
          "hide_name": 0,
          "type": "$assert",
          "parameters": {
          },
          "attributes": {
            "hdlname": "spi_axi_controller_inst spi_axi_sms_inst 679"
          },
          "port_directions": {
            "A": "input",
            "EN": "input"
          },
          "connections": {
            "A": [ "1" ],
            "EN": [ "1" ]
          }
        },
```

Is this expected? Is this something you'd expect the ghdl plugin to handle or something you'd expect yosys to strip downstream? I know there are varying expectations around what different standard EDA tools might do with this case.

In my case, this is easily worked around by dumping the assert that no longer provides much value.

tgingold, 2025-02-12:
Either use the `--no-formal` option to the ghdl plugin, or use the yosys pass which removes assert cells.

If this cell is still present after optimization, it means that yosys was not able to remove it.  Maybe it is not easy to prove this state is never reached.

nathanaelhuffman, 2025-02-12:
--no-formal works for my usecase thanks!

closed 2025-02-12 by nathanaelhuffman