ghdl/ghdl-yosys-plugin #211: mingw-w64-x86_64-yosys package does not contain plugins/ghdl.so file

JakubFranek, 2025-02-15:
Hello,

I installed `mingw-w64-x86_64-yosys` package from the `mingw-w64-x86_64-eda` package group via MSYS2 on my Windows 10 machine as recommended in the readme. 

When I try to show a schematic view of a design via TerosHDL, which invokes `ghdl-yosys-plugin`, I get the following error:

```
2025-02-16 00:14:00.905 [info] Executing: C:\WINDOWS\system32\cmd.exe  C:\WINDOWS\system32\cmd.exe /d /s /c " D:\Programy\MSYS2\mingw64\bin\yosys.exe -m ghdl -p "ghdl --std=08 -fsynopsys  --work=work "d:\Electronics\FPGA\nexys-a7-lab\source\common\clock_enable_generator.vhd"  --work=work -e clock_enable_generator; hierarchy -top clock_enable_generator; proc; write_json C:\Users\jfran\.teroshdl_oz0Bm; stat""
2025-02-16 00:14:01.599 [info] ERROR: Can't load module `./ghdl': "D:\Programy\MSYS2\mingw64\bin\..\share\yosys\plugins\ghdl.so": Uveden� modul nebyl nalezen.
```

(The ending of the second error line is in Czech and means "listed module not found").

Indeed I can confirm there is no `plugins` folder within my `MSYS2/mingw64/share/yosys/` folder and no `ghdl.so` file anywhere within my whole MSYS2 directory.

According to [this Reddit discussion](https://www.reddit.com/r/FPGA/comments/17di125/comment/ksw5vw2/), I am not alone in encountering this exact same error. Is it possible that the recommended installation procedure as listed in this repository's readme is out of date or invalid? It seems to me that the recommended pre-built package does not actually contain this plugin after all.

Thank you.

Kind regards
Jakub

tgingold, 2025-02-16:
I am not sure what should be fixed.  Now yosys supports plugin on Windows, so I will modify the README for that.

But the README simply claims that *some* projects provide ghdl-yosys-plugin.
On mingw, you still have to build it.

tgingold, 2025-02-16:
I agree it is not very well worded.   I will improve that.

std-rubic, 2025-02-21:
Hello i have the Same problem. And added the issue in

https://github.com/msys2/MINGW-packages/issues/23466

Kreijstal, 2025-02-26:
.so files are linux files, you mean .dll?