ghdl/ghdl-yosys-plugin #212: Fedora 41 - install ghdl-yosys-plugin - error in library path

lionelsaintecluque, 2025-03-04:
Hello, 

I'm trying to install ghdl-yosys-plugin. I run fedora 41 and I got ghdl with dnf. 

The makefile of ghdl-yosys-plugin uses the following command to get the GHDL library path : 
`LIBGHDL_LIB:=$(shell $(GHDL) --libghdl-library-path)`
which results in : /usr/lib/libghdl-3_0_0.so

When I make it, the linker complains because it does not finds the library. 

when I run ls, and find to check the presence and location of the library, I don't find it in /usr/lib, but in /usr/lib64 : 
`ls /usr/lib*/libghdl*
/usr/lib64/libghdl-3_0_0.so  /usr/lib64/libghdlvpi.so`

Is there a bug in the distro ? Or something I did bad ?

sebeaumont, 2025-03-11:
I'm just going through this torture myself. So far I have had to build `ghdl` from source as the Fedora packages are bogus and do not include the c headers. So check which version of `ghdl` you are using (see below) I removed all the Fedora packages:
```
$ which ghdl
/usr/local/bin/ghdl 
$ ghdl --version
GHDL 6.0.0-dev (4.1.0.r598.ge58e803ec) [Dunoon edition]
 Compiled with GNAT Version: 14.2.1 20250110 (Red Hat 14.2.1-
 static elaboration, mcode JIT code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2025 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
This will at least get you all the required ghdl in ```/usr/local```
```
$ ghdl --libghdl-library-path
/usr/local/lib/libghdl-6_0_0_dev.so
$ ghdl --libghdl-include-dir 
/usr/local/include
```
You will need  `gcc-gnat` to install `ghdl` and if you install libraries in `/usr/local` then you will have to add an `ldconfig` rule to make sure that `/usr/local` is searched. 

`ghdl-yosys-plugin` also needs `readline-devel` and also `tcl-devel`
At which point assuming all paths are correct (**one to watch for is that `oss-cad-tools` provides a ghdl binary but no libraries or headers so make sure the one you install is before that on the search path**). You should be able to get `make` to build you a `ghdl.so`  (not a great name for the plugin shared library IMO but there you go WIP).
```
ldd  ghdl.so  
        linux-vdso.so.1 (0x00007f1b1c8be000)
        libghdl-6_0_0_dev.so => /usr/local/lib/libghdl-6_0_0_dev.so (0x00007f1b1c200000)
        libstdc++.so.6 => /lib64/libstdc++.so.6 (0x00007f1b1be00000)
        libm.so.6 => /lib64/libm.so.6 (0x00007f1b1c11a000)
        libreadline.so.8 => /lib64/libreadline.so.8 (0x00007f1b1c81d000)
        libffi.so.8 => /lib64/libffi.so.8 (0x00007f1b1c80c000)
        libz.so.1 => /lib64/libz.so.1 (0x00007f1b1c7e6000)
        libtcl8.6.so => /lib64/libtcl8.6.so (0x00007f1b1bc5c000)
        libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f1b1c0eb000)
        libc.so.6 => /lib64/libc.so.6 (0x00007f1b1ba69000)
        libgnat-14.so => /lib64/libgnat-14.so (0x00007f1b1b400000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f1b1c8c0000)
        libtinfo.so.6 => /lib64/libtinfo.so.6 (0x00007f1b1c0bd000)
```  
Quite important to make sure the dependencies make sense here. Then this shared object must be copied to the right place using `make install`.
```
make install
yosys-config --exec mkdir -p /home/seb/.osscad/oss-cad-suite/share/yosys/plugins
yosys-config --exec cp ghdl.so /home/seb/.osscad/oss-cad-suite/share/yosys/plugins
```
YMMV  there -- good luck!

sebeaumont, 2025-03-11:
I'm now going to try building the `llvm` based backend for `ghdl` so I can use it on `aarch64` Fedora41.

lionelsaintecluque, 2025-03-11:
Hello, 

I'm surprised GHDL does not works well for you in fedora. For me it's good. I'm running fedora 41 (with a kernel of Fedora 38, to be compatible with my graphic chipset, but it's irrelevant for what we are discussing I think). 

I edited the Makefile of ghdl-yosys-plugin like this : 
LIBGHDL_LIB:=$(shell $(GHDL) --libghdl-library-path)
#LIBGHDL_LIB=/usr/lib64/libghdl-3_0_0.so

So far it works. 

My main concern is that `ghdl --libghdl-library-path` does not returns the right location. It can generate troubles to other users. Perhaps I should raise an issue in GHDL project, rather than here.

sebeaumont, 2025-03-11:
`ghdl` returns the right paths -- maybe you didn't check `$which ghdl`

sebeaumont, 2025-03-11:
I think the problem is with the Fedora package. I have just checked this on another system with the ghdl package and get the same results as you:
```
$ which ghdl                      
/usr/bin/ghdl
$ ls $(ghdl --libghdl-library-path)
ls: cannot access '/usr/lib/libghdl-3_0_0.so': No such file or directory
```
As I said I'm steering clear of the Fedora packages.

lionelsaintecluque, 2025-03-11:
For ghdl-yosys-plugin installation issue I think we can close:

- You provided in detail way to do with one method.
- I provided another way of doing it by editing Makefile. 

Thanks a lot for the information you shared.

closed 2025-03-11 by lionelsaintecluque